OpenHD/OpenHD #411: Collection of harware that might be usable for OpenHD

Consti10, 2021-01-25:
Just paste a link here, and maybe some more info

Consti10, 2021-01-25:
https://www.cnx-software.com/2021/01/22/rockchip-rv1126-ai-camera-soc-features-2-0-tops-npu-promises-250ms-fast-boot/

http://www.thinkcore.cn/product/detail/23/35

Consti10, 2021-01-25:
https://www.cnx-software.com/2021/01/25/firefly-dual-lens-ai-camera-module-comes-with-rockchip-rv1109-or-rv1126-processor/

Consti10, 2021-01-25:
https://www.crowdsupply.com/luxonis/megaai

Consti10, 2021-01-25:
https://de.aliexpress.com/item/32777314627.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=40018053811&tt=MG&aff_fsk=_A6mA8z&aff_platform=default&sk=_A6mA8z&aff_trace_key=160533bb2d2e4c3db42551390f4b86d4-1611607325930-05937-_A6mA8z&shareId=40018053811&businessType=ProductDetail&platform=AE&terminal_id=28782b2d45774e76af39581f1e84a055

https://de.aliexpress.com/item/32960113528.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=40018195675&tt=MG&aff_fsk=_AXvIP3&aff_platform=default&sk=_AXvIP3&aff_trace_key=ae7f237ff8b542dfb81266c23a3ea177-1611607338605-04328-_AXvIP3&shareId=40018195675&businessType=ProductDetail&platform=AE&terminal_id=28782b2d45774e76af39581f1e84a055

Consti10, 2021-01-25:
https://de.aliexpress.com/item/32960113528.html?srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=40018195675&tt=MG&aff_fsk=_AXvIP3&aff_platform=default&sk=_AXvIP3&aff_trace_key=ae7f237ff8b542dfb81266c23a3ea177-1611607338605-04328-_AXvIP3&shareId=40018195675&businessType=ProductDetail&platform=AE&terminal_id=28782b2d45774e76af39581f1e84a055
same as:

https://www.uctronics.com/720p-gc1024-h-264-usb-camera-uvc-compliance.html

Consti10, 2021-01-25:
tested by zipray, ~170ms end to end on ez-wb:
https://de.aliexpress.com/item/4000512924177.html?spm=a2g0n.detail.0.0.5f202229KEOe04&gps-id=storeRecommendH5&scm=1007.18500.187585.0&scm_id=1007.18500.187585.0&scm-url=1007.18500.187585.0&pvid=44675685-59f3-4936-84ec-af243095a5f1&_t=gps-id%3AstoreRecommendH5%2Cscm-url%3A1007.18500.187585.0%2Cpvid%3A44675685-59f3-4936-84ec-af243095a5f1%2Ctpp_buckets%3A668%230%23131923%2310_668%23888%233325%235_668%232846%238107%231934_668%232717%237565%23771__668%233374%2315176%23839&browser_id=5c3ba34bd1914299b3d01b1407157bc1&aff_trace_key=b8d5b15ce70742f5a08ee149d2e93123-1608988608133-04046-UneMJZVf&aff_platform=msite&m_page_id=bcpxf1friiqcaqfx1773446272bc9705a6022051e1&gclid=&_imgsrc_=ae01.alicdn.com%2Fkf%2FHd751074d157f4dc3bab55149ff78a672X.jpg_640x640Q90.jpg_.webp

CopterGUI, 2021-01-26:
https://www.mouser.de/ProductDetail/Silex-Technology/SX-NEWAH-SP-US?qs=sGAEpiMZZMsjLMBIknjmknKJE5m7ZCsmcTd9mrLeZQhEupQKrLXyHA%3D%3D

900mHz wifi modules

CopterGUI, 2021-04-01:
https://www.karo-electronics.com/de/qsxp

Consti10, 2021-04-01:
RV1126:
https://item.taobao.com/item.htm?spm=a230r.1.14.53.6cf7407aGBl8Pd&id=637071844357&ns=1&abbucket=6#detail

Consti10, 2021-04-01:
Rn there is
RPI CM4
Jetson nano
RV1126 module (https://item.taobao.com/item.htm?spm=a230r.1.14.42.2698407al3BRUL&id=632843493425&ns=1&abbucket=6#detail)
IMX module (https://www.karo-electronics.com/de/qsxm)

1)RPI CM4 has great software support and can do 1080p30 / 720p60. It is also increadibly small and cheap

2)Jetson nano unfortunately is quite big due to its interconnect, but also priced competitvely, and has increadibly good encoding capabilities.

3)both RV1126 and IMX have (on paper) pretty much the same encoding capabilities, somewhere between rpi and jetson
but the RV1126 seems to be a bit cheaper, and readily available as "module" on taobao

e.g. the RV1126 carrier sells for ~34.27€ on taobao, that's hard to beat. Pretty much the same as CM4 but better

Consti10, 2021-04-04:
RV1129 specs read:
h264 level 5.1
h265 level 5.0
RV1109 doesn't specify level for some reason,
but they say (excerpt)
RV1126 does
3840x2160 @ 30 + 1080p @ 30 (2 streams)
Whereas RV1109 only has:
3072x1728 @ 30 + 1280x720p @ 30
So it seems as if RV1126 has slightly better encoding capabilities.
Normally, you can swap resolution for framerate and other way around.
But strictly speaking, only level 5.1 allows for
1920x1080@120fps 
Which is really good
since they are really similar in price, RV1126 is probably better for us
even though we don't need the CPU cores or NPU

RV1126 module board(s) suitable for custom carrier:
1)https://item.taobao.com/item.htm?spm=a230r.1.14.69.6cf7407a2LJuq8&id=640762445304&ns=1&abbucket=6#detail
2)https://item.taobao.com/item.htm?spm=a1z10.5-c-s.w4002-22747001949.57.67251e61s6UnW6&id=641752963533

Consti10, 2021-04-25:
List of manufacturers for SOCs with integrated h264/h265 encoder:
[Rockchip](https://www.rock-chips.com/a/en/products/RV11_Series/2020/0427/1076.html)
HiSilicon
[Ingenic](http://www.ingenic.com.cn/en/?product/id/33.html)

macrokernel, 2021-05-05:
Not sure about this SoM, priced at around 30 USD, based on NXP IMX6ULL, 42.35 x 45.55 mm, boasts CSI support: 
https://m2m-tele.com/product/netsom/

closed 2023-02-23 by Consti10