OpenHD/OpenHD #412: When is OpenHD expected to support raspberry pie CM4, thank you!

wqq3000, 2021-01-26:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration and debug logs**
Please attach the relevant openhd-settings*1.txt and any debugging logs available

**Hardware:**
 - Air side
 -- Pi model
 -- Camera(s)
 -- Wifi card(s)
 -- Other hardware
 - Ground side
 -- Pi model
 -- Wifi card(s)
 -- Ground Control Station(s)
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?

**Client devices (in case of hotspot/tethering issue)**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App - FPV_VR, QGC, and version
 - Version [e.g. 22]
 - Connection method

**Additional context**
Add any other context about the problem here.

CopterGUI, 2021-01-29:
It does work already with the official io board.

Just add this to the config.txt

"dtoverlay=dwc2,dr_mode=host"

CkovMk, 2021-07-20:
> It does work already with the official io board.
> Just add this to the config.txt
> "dtoverlay=dwc2,dr_mode=host"

@CopterGUI  It seems CM4 doesn;t work for me either... Could you plz help me out here? (#460 )

closed 2022-10-17 by raphaelscholle