OpenHD/OpenHD #416: [Feature Request] Multiple USB Cameras and a USB Camera as primary Camera

MCMH2000, 2021-02-06:
Hello, I want to use my Actioncam as the primary Camera and an HT-301 Thermal Camera as the second Camera on the AirPi,
so i added an HDMI to USB Capture Card for the actioncam and connected the thermal camera simply via USB.
The capture card is listed as /dev/video0 and the thermal as /dev/video1, so i wanted to configure the AirPi for these two USB Devices, so that when I boot, i see the Actioncam's Video and can switch to the thermal camera with my Controller. But there is no way I can do that? Please correct me if i'm wrong, but when I try to boot the AirPi without any CSI Camera attached it won't be establish a link to my GroundPi although two video devices are connected to the AirPi.

I thought the AirPi recognizes when a video device is connected and will configure it self as the AirPi.

I don't want to buy me a Raspberry Pi Cam just to use Open.HD.
And apart from this, is there any possibility to use two USB video devices at all?
The only thing I can do is, to configure one single USB video device as the second camera. Why is there no option to configure the primary camera?

ZeroUno87, 2021-02-14:
There is a procedure to force the Pi as air device. Read the documentation, it’s clearly explained. Look for USB camera

CopterGUI, 2021-02-15:
They way to go here is to use a HDMI-CSI adapter instead of the hdmi-usb capture card. Most likely your capture card does not output h264.

MCMH2000, 2021-02-15:
> There is a procedure to force the Pi as air device. Read the documentation, it’s clearly explained. Look for USB camera

Yes, i finally found it in the gitbook.io docs, which were btw not linked in the github wiki.
But when i force the Pi with the air.txt then i am still only able to use one single usb cam.
Therefore it would be cool if something like multiple usb cameras were possible.

MCMH2000, 2021-02-15:
> They way to go here is to use a HDMI-CSI adapter instead of the hdmi-usb capture card. Most likely your capture card does not output h264.

That is what i have done now. It is working but now I have trouble to switch between the two cameras.
In https://openhd.gitbook.io/open-hd/advanced-setup/using-an-ip-or-usb-camera-as-second-camera 
is a note at the bottom: "Please be aware you can switch between the feeds from the main QOpen.HD interface as well without the need of a joystick connected to the Ground SBC." but how?
Is there a function in the QOpen.HD interface, do i have to connect a keyboard or a mouse to the GroundPi or is it possible to control the video feed via touch?

And for clarify my understanding: The AirPi is streaming both camera feeds to the GroundPi and the switching between the feeds is then only done by the GroundPi, correct?
And if this is correct, are both feeds available as a stream at the same time, so that i can integrate them in OBS? Because the primary and secondary stream have different ports. Or is there just one stream whose feed is controlled by the GroundPi?

ZeroUno87, 2021-02-15:
QOpen.HD supports picture in picture function. You have to enable it in the QOpen.HD app settings. It seems to properly work. never tried att

ZeroUno87, 2021-02-15:
Anyway I suggest you to enter in telegram channel ( https://t.me/OpenHD_HDFPV )... it's very active and there are a lot of developers a people with a lot of experience that are happy to help all of us.... Theoretically the 'issues' on GitHub are done to collect bugs and common request.
You can also take a look to the public forum! ( https://discuss.openhdfpv.com/ )

MCMH2000, 2021-02-15:
> Anyway I suggest you to enter in telegram channel ( https://t.me/OpenHD_HDFPV )... it's very active and there are a lot of developers a people with a lot of experience that are happy to help all of [...]

Yes, I will outsource the topic to the forum, sorry for asking this many questions here, you are right.
But anyway, thank you for the help.

ZeroUno87, 2021-02-15:
Hey Bro! Don’t ask sorry! My post wasn’t a complain!!! I wanted to say that asking in the other places it will be much more easy and fast to get the right solution! Enjoy with OHD! 🤙

raphaelscholle, 2023-01-31:
2.3 doesn't care what camera type is primary or secondary

closed 2023-01-31 by raphaelscholle