OpenHD/OpenHD #423: video recording on airpi

JacoJefferson, 2021-03-14:
Hello friends. All right with you? Hope so. I would like to know if it is possible to record video from the pi camera, for example, in 1080p quality, on the airpi sd card itself, at the same time that 720p video transmission is possible, for example, to the groundpi, in order to have video recorded with quality and a light fpv?

steveatinfincia, 2021-03-14:
It is but there are limitations. The older 2.x releases have an experimental option called openhdvid that could do what you’re describing but it was just for testing, and on a pi if you stream a lower resolution the pi will have to resize it real-time and that adds latency (not a lot but it’s noticeable).

The unreleased 3.x branch can record on both sides but it’s still going to have that resizing limitation on the pi, so we will likely not have it resize by default and just stream the same video that’s being recorded.

It’s possible that other platforms might behave differently, and in general there’s not much to be gained by reducing resolution due to the way h264 works, if you’re hoping to get reduced bitrate it won’t be a significant difference.

JacoJefferson, 2021-03-14:
Got it. I appreciate the explanation. I would just like to add that it would also be useful for continuous video recording, without the flaws inherent to live streaming. thank you once again and congratulations on the project.

hyperion11, 2021-03-17:
for recording USB SSD can be used, i think. Especially on pi4 with USB3.0

steveatinfincia, 2021-03-17:
Yes an SDD is probably smart, even usb2 is enough as long as the SSD doesn’t wait a lot and tie up the bus

LimitlessGreen, 2021-11-06:
This issue has been mentioned on **OpenHD**. There might be relevant details there:

https://forum.openhdfpv.org/t/record-videos-on-airpi/583/2

raphaelscholle, 2023-01-31:
Air recording ins enabled in 2.3

closed 2023-01-31 by raphaelscholle