OpenHD/OpenHD #433: Image builder fails for Jetson Nano

jhw84, 2021-04-08:
When I try to use the image builder (from master) to generate an image for the Jetson Nano devboard:
`./build.sh jetson-nano-4gb-bionic`
it gives the following error during one of the stages:

> [11:16:29] Begin /Open.HD_Image_Builder/stages/02-Packages/00-run-chroot.sh
Binding host partitions on /Open.HD_Image_Builder/work-jetson-nano-4gb-bionic/02-Packages/mnt
Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:5 https://repo.download.nvidia.com/jetson/common r32.4 InRelease [2552 B]
Ign:6 https://repo.download.nvidia.com/jetson/t120 r32.4 InRelease
**Err:7 https://repo.download.nvidia.com/jetson/t120 r32.4 Release**
  **404  Not Found [IP: 2.22.54.120 443]**
Get:8 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 Packages [1228 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main Translation-en [402 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports bionic-updates/restricted arm64 Packages [3052 B]
Get:11 http://ports.ubuntu.com/ubuntu-ports bionic-updates/restricted Translation-en [38.2 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 Packages [1518 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe Translation-en [366 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports bionic-updates/multiverse arm64 Packages [4912 B]
Get:15 http://ports.ubuntu.com/ubuntu-ports bionic-updates/multiverse Translation-en [6464 B]
Get:16 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 Packages [930 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports bionic-security/main Translation-en [310 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports bionic-security/restricted arm64 Packages [2480 B]
Get:19 http://ports.ubuntu.com/ubuntu-ports bionic-security/restricted Translation-en [34.6 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 Packages [975 kB]
Get:21 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe Translation-en [251 kB]
Get:22 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse arm64 Packages [2720 B]
Get:23 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse Translation-en [4412 B]
Get:24 https://repo.download.nvidia.com/jetson/common r32.4/main arm64 Packages [14.0 kB]
Reading package lists...
**E: The repository 'https://repo.download.nvidia.com/jetson/t120 r32.4 Release' does not have a Release file.**

I am running the builder on Ubuntu 20.04. I tried also in a Docker container in Ubuntu 18.04, which gives the same result.

wx4cb, 2021-04-08:
check the nvidia repository for the jetson. it cannot find the nvidia libraries that it needs in apt when it;'s doing an apt-update.

is there a bionic repo for the jetson ?

when i try to go to that repo myself i get the following, so I would look up on the nvidia site for the jetson to see if they moved it.

```
This site can’t be reachedThe webpage at https://repo.download.nvidia.com/jetson/t120 might be temporarily down or it may have moved permanently to a new web address.
ERR_INVALID_RESPONSE

```

steveatinfincia, 2021-04-08:
I think it’s a typo in the base image, it should be t210 rather than t120 I’ll need to repack those base images today. I think only one of them is affected, the 2gb should be ok.

jhw84, 2021-04-08:
> I think it’s a typo in the base image, it should be t210 rather than t120 I’ll need to repack those base images today. I think only one of them is affected, the 2gb should be ok.
> […](#)

Just did a quick check, the 2gb seems to have the exact same typo (t120 instead of t210). When just trying to fetch the url in a browser (as @wx4cb did) both t120 and t210 don't work, so perhaps nvidia moved some things around as well.

jhw84, 2021-04-08:
> > I think it’s a typo in the base image, it should be t210 rather than t120 I’ll need to repack those base images today. I think only one of them is affected, the 2gb should be ok.
> > […](#)
> 
> Just did a quick check, the 2gb seems to have the exact same typo (t120 instead of t210). When I change to t210 in the base image, repack and run the builder this step now works fine.
> However, a different error occurs later on: `E: Unable to locate package openhd`. I will double check that I didn't mess anything up when changing the apt/sources list

Never mind, I forgot to add the "testing" argument to the build script.
Everything works fine now, so all that had to be changed was this url in the apt sources list in the base image (t120->t210)

raphaelscholle, 2021-04-12:
yay something I contributed ;)

closed 2022-10-17 by raphaelscholle