OpenHD/OpenHD #434: Can i use old osd in Buster firmware?

rebelboy, 2021-04-11:
Can i use old osd in Buster firmware?

My Ground pi :  4B
              air pi :  zero w

careyer, 2021-04-11:
No, this is not possible. Stretch uses a deprecated graphics library to render the old osd. That library is not available with Buster anymore.

rebelboy, 2021-04-11:
Thanks!

The new OSD font is too small.
My monitor is 7 inches (feelworld fw279).

I can't see the battery Voltage easily from picture1.
but I can see the voltage and current easily from picture2 (old osd).

[image]
[image]

pilotnbr1, 2021-04-11:
Tap (if you have touch) or click the widgets you want to change. There is a
"size" slider...

closed 2021-07-09 by pilotnbr1