OpenHD/OpenHD #440: WiFi cards not detected if Taranis USB connected during boot

TheGuv, 2021-05-03:
If my Taranis radio (which I use for RC control via USB HID) is left plugged into any of the Ground Pi's USB ports during boot up, neither of the two attached ALFA AWUS036NHA WiFi cards are detected as shown by the message 'No wifi card found' on an attached HDMI display. If I wait until the WiFi cards are detected (on these, there is a blue LED indicator that illuminates) and then plug in the Taranis, all is well

openhd-settings*1.txt is unmodified from defaults.

Ground Pi is a 3B+
Wifi as noted ALFA AWUS036NHA
Taranis is QX7

wx4cb, 2021-05-03:
do you have the taranis set to HID or  or usb storage ?

CopterGUI, 2021-05-03:
You can try and re-arrange the usb port order to make it work.

TheGuv, 2021-05-03:
> do you have the taranis set to HID or or usb storage ?

It is set on HID. 
It's not an issue with the Taranis working as joystick but rather when it is plugged in at boot time, the WiFi cards are not detected.

TheGuv, 2021-05-03:
> You can try and re-arrange the usb port order to make it work.

As in, which physical ports things are plugged into. OK, will try. Thanks.

closed 2023-01-31 by raphaelscholle