OpenHD/OpenHD #447: [Feature Request] CRSF to Ground

CorruptName, 2021-06-08:
What I'm after:
I would like to connect Crossfire Receiver to Ground via CRSF or Mavlink serial protocols and let OpenHD still handle the Video, RC, and Telemetry from Air to Ground. This way I I can connect to the ground station from tx and get telemetry back on the TX. The reason is so that I can fly without Mission Planner or QGC or  even goggles if I choose and will still have telemetry on Radio.
Thanks and I'm loving the work.
BTW, I also have a mostly completed python library for using BST so the receiver could almost be able to communicate with the Ground station using I2C for those who aren't using Pi4 or only have 1 UART.

wx4cb, 2021-06-08:
Are you planning on releasing the library?

CorruptName, 2021-06-08:
> Are you planning on releasing the library?

After looking at where I left the library I realized that it is in cpp and not python. That's why I deleted that part from my comment above. But to answer your question, the answer is yes. Just don't have as much time these days with a new baby and another on the way.

CorruptName, 2021-06-08:
> Are you planning on releasing the library?

Dang you are everywhere lol, You commented on my post about INAV CRSF Telemetry.
Just Trolling me?

wx4cb, 2021-06-08:
> > Are you planning on releasing the library?
> 
> Dang you are everywhere lol, You commented on my post about INAV CRSF Telemetry.
> Just Trolling me?

hahahaha  - i'm like a bad penny :D

closed 2023-01-31 by raphaelscholle