OpenHD/OpenHD #464: Banana Pi Zero Support? (BPI-M2)

CorruptName, 2021-08-08:
I'm curious if this SBC can be used? It features the same footprint as the raspberry pi zero but has a AllWinner H2+ quad-core 1GHz (oc to 1.2GHz with cooler) and a Mali-600 GPU. Cost around $25USD ATM. Seems like it would be an excellent replacement for RPI Zero? Says it supports Raspbian. Do you think we could make it work with OpenHD?

white-rabbitGR, 2021-08-08:
Currently only Raspberry Pi SBCs are supported and a very rare and hard to buy 3B mini SBC that has the same CPU of RPI3B. Please have a look here for information regarding this board:

https://youtu.be/B4U8i7FIbfA

CorruptName, 2021-08-08:
Not really interested in the "3b mini" - was already subb'd to MarioFPV. I
was looking at alternative "real" products.
Do you actually know the BPI-M2 won't work? Or are you just telling me that
you read the "supported boards" section like everyone else? Thanks for your
help?

pilotnbr1, 2021-08-08:
> I'm curious if this SBC can be used? It features the same footprint as the raspberry pi zero but has a AllWinner H2+ quad-core 1GHz (oc to 1.2GHz with cooler) and a Mali-600 GPU. Cost around $25USD [...]

"Make it work"? Sure. You will need to write some code

Will it work with the official 2.0.8 release?no
Does any image builder version target that board?no

There also is no current interest in targetting  that board. Wont go into reasons why... Search the telegram thread, rcgroups, or Constantines repo/posts for the "why". Current interest in non pi sbc are the jetson nano and the rv1126. Costum compute module boards also have had interest.

CorruptName, 2021-08-08:
> "Make it work"? Sure. You will need to write some code
>
> Will it work with the official 2.0.8 release?no
> Does any image builder version target that board?no
>
> There also is no current interest in [...]

Thanks for your input.
I don't really care what arbitrary reason "why" (not entirely sure "why" we're putting that in quotes) but it was a simple question of how complicated supporting this board would be (officially or unofficially - at the end of the day, it's just linux). I wasn't really asking for people to come and tell me what the supported boards were. Perhaps I will just order a couple and build an image.

closed 2021-08-08 by pilotnbr1