OpenHD/OpenHD #465: Disk image contains a large amount of undeletable files

rsaxvc, 2021-08-09:
**Describe the bug**
The compressed release disk images(ex: Open.HD-2.0.8-buster.img.gz ), are quite a bit(~1/4-1/3) larger than necessary due to undeletable files and  presence of a large amount of non-zero slack space.

This isn't really a bug report but an improvement suggestion.

**To Reproduce**
Steps to reproduce the behavior:
1. Download Open.HD-2.0.8-buster.img.gz and uncompress
2. Use `testdisk Open.HD-2.0.8-buster.img`, select proceed, select 'Intel' partition type, select "image creation" on partition 2(rootfs), press c to extract to image.dd to analysis
4. To generate a reference of the compressed partition, 'gzip -k image.dd', and rename to to keep around 'mv image.dd.gz image_reference.dd.gz'
5. to zero the slack space, run 'zerofree -v image.dd'
6. compress again with 'gzip -k image.dd' and rename output 'mv image.dd.gz image_zerofreed.dd.gz'
7. evaluate size difference with 'du -h image_*.dd.gz'
8. reassemble image and verify operation

**Expected behavior**
Would be nice if disk image compressed better than the current releases do.

**Configuration and debug logs**
Stock 2.0.8 release image

**Hardware:**
 - SD Release Image

**Client devices (in case of hotspot/tethering issue)**
 - None involved

**Additional context**
Just noticed the releases could compress better.

It looks like the image builder's scripts/common.sh unmount_image function could probably do this automatically.

raphaelscholle, 2023-01-31:
in 2.3 we use .xz

rsaxvc, 2023-02-20:
That's not quite what I mean, and xz certainly helps, but I mean that before compression, the image is ~6.6GB, but only contains ~4.8GB of useful data(based on building 2.3-evo myself, and running PiShrink), and a bunch of leftover build bits in the slack space.

LimitlessGreen, 2023-02-20:
I can look into it. Maybe it's possible to integrade PiShrink into the build process.

raphaelscholle, 2023-02-22:
I tweaked the image, now it is 3.88 GB before compression

closed 2023-02-23 by raphaelscholle