OpenHD/OpenHD #483: 2.1 Dev - Hotspot data not shared/readable in qopenhd

pilotnbr1, 2021-09-14:
Hotspot is created by 2.1 groundpi however when connecting to it via linux laptop running qopenhd, no telemetry is reflected on the app.

pilotnbr1, 2021-12-21:
Telemetry is now working on groundpi but still not distributed to connected devices- nothing displayed on on qopenhd app running on linux connected to groundpi via wifi

patemil, 2022-02-02:
I think this is partially solved now, still missing ground/air load and wifi status

pilotnbr1, 2022-02-03:
I closed this a bit prematurely.. As @patemil said theres a bit more work. 

I think we should resolve this with extended protocol sending off updates. I think air/ground load and wifi rssi should be fairly slow updating. This is not data to burn up alot of air time over

pilotnbr1, 2022-02-24:
Tested and seems to be working now! Good job @patemil

closed 2022-02-24 by pilotnbr1