OpenHD/OpenHD #484: 2.1 Dev - openhd_interface error

pilotnbr1, 2021-09-14:
Code errors after setup_hotspot

WiFi::setup_hotspot()
Error: locale::facet::_S_create_c_locale name not valid

pilotnbr1, 2021-09-16:
Only happens on Pi

pilotnbr1, 2021-12-23:
"export LC_ALL=C" in terminal fixes error but that needs to be permanent or determined why that works

pilotnbr1, 2021-12-23:
This error does not happen when the code is run right on a pi.. It only happens when ssh in. This leads me to believe that during ssh my desktop terminal is passing locale variables that the pi does not recognise.

closed 2021-12-23 by pilotnbr1

pilotnbr1, 2022-05-13:
@Consti10 reports:

locale issue is generated by 
https://github.com/OpenHD/Open.HD/blob/consti-test/OpenHD/ohd_common/openhd-util.hpp#L25

So unless it is not existing, every single command run by stephen in his services might fail for dubious reasons

Consti10, 2022-05-13:
> @Consti10 reports:
> 
> locale issue is generated by https://github.com/OpenHD/Open.HD/blob/consti-test/OpenHD/ohd_common/openhd-util.hpp#L25
> 
> So unless it is not existing, every single command run by stephen in his services might fail for dubious reasons

Correction: at least when you run it manually via ssh ;) (going by your ssh observation)