OpenHD/OpenHD #486: DISPLAY_OSD=NO results in black screen

elRadish, 2021-09-21:
**Describe the bug**
When DISPLAY_OSD=NO in the settings file is set, the HDMI output is entirely black.
When it is set to =YES, Video and OSD are shown.

**To Reproduce**
Steps to reproduce the behavior:
1. Set DISPLAY_OSD=NO in openhd-settings-1.txt
2. Start both sided
3. Only a black screen is shown, no video

**Expected behavior**
Video output without OSD

**Configuration and debug logs**
[openhd-settings-1.txt](https://github.com/OpenHD/Open.HD/files/7204095/openhd-settings-1.txt)

**Hardware:**
 - Air side
 -- RP 0
 -- Raspberry default cam
 -- AC56
 -- Other hardware
 - Ground side
 -- RPI4
 -- AC56

leohku, 2021-09-24:
What version of OpenHD are you running on the ground and air?

leohku, 2021-09-24:
If you're running v2.0, I believe the description `DISPLAY_OSD` in `openhd-settings-1.txt` is incorrect. Instead of setting "N" to only display video, it will actually cause neither QOpenHD nor the Wifibroadcast OSD to display, resulting in a black screen.

See [this link](https://github.com/OpenHD/Open.HD/blob/2.0/wifibroadcast-scripts/osd_rx_functions.sh#L61-L67) for the code that causes this behaviour.

leohku, 2021-09-24:
@pilotnbr1 Do you think the description of `DISPLAY_OSD` should be edited correctly reflect its behaviour? I can submit a PR if needed.

From:
```
# Set to "Y" to enable ground OSD. set "N" to only display video. 
```

To:
```
# Set to "Y" to enable ground OSD. set "N" to disable all video output on screen.
```

MacDaddyFPV, 2021-09-27:
> @pilotnbr1 Do you think the description of `DISPLAY_OSD` should be edited correctly reflect its behaviour? I can submit a PR if needed.
>
> From:
>
> ```
> # Set to "Y" to enable ground OSD. set "N" to only [...]

That would be great thanks.

pilotnbr1, 2021-09-29:
The old intended behavior is display_osd=N should result in just video on the pi and no osd. This was a throw back to stretch and the old openvg osd which is now depreciated. Under qopenhd users have the option to just disable all widget thereby creating a blank screen.

I think this option should just be removed all together because it doesnt work and is superceded by the options within the qopenhd app

MacDaddyFPV, 2021-09-29:
Perhaps it would be handy if we replace this option with a switch in QOpenHD that can disable all widgets so that a user does not have to click through each one individually and then remember what they had on when switching them back on.

pilotnbr1, 2021-09-29:
Great idea

leohku, 2021-09-30:
I can try to work on removing the `DISPLAY_OSD` option from the OpenHD repo, and see if I can add a switch in QOpenHD as per suggestion. Will be a good first issue for me.

MacDaddyFPV, 2021-09-30:
> I can try to work on removing the `DISPLAY_OSD` option from the OpenHD repo, and see if I can add a switch in QOpenHD as per suggestion. Will be a good first issue for me.

Sounds excellent. If not already then please jump on telegram and we can add you to the QOpenHD Dev group.

closed 2023-01-31 by raphaelscholle