OpenHD/OpenHD #489: 2.0.12b PROBLEMS

pilotnbr1, 2021-09-23:
-Android adsb airplane icon missing.. Probably font library problem.

-For mission waypoints on map- lines connecting waypoints sometimes not in correct order.

paulick, 2021-09-25:
Nice to see that development is going on!
I started the new image and the air unit is nicely comming up but groundstation stopped initialization with "no wificard" and no led comming up on wlan adapter. i can't get a console with ctrl+F12 or ssh/putty.

airpi:
-raspi cm3
-awus036ACH (rtl8812au)

groundpi:
-raspi 4b
-awus036ACH (rtl8812au)

i flashed openhd 2.0.8 on ground again and everything is fine (except ethernethotspot not forwarding videostream to a windows machine)
2.0.8 and 2.0.12 working nicely together. no errors and a stable videostream to groundpi.

pilotnbr1, 2021-09-25:
Thanks paulick!

MacDaddyFPV, 2021-09-27:
> Nice to see that development is going on!
> I started the new image and the air unit is nicely comming up but groundstation stopped initialization with "no wificard" and no led comming up on wlan [...]

Hi Paulick. Someone has the same setup which is working. Could you please contact me (Daniel McCullock) in telegram and I can help troubleshoot the issue?

reichley, 2021-09-27:
My apologies but just opened an issue to the same effect. Any resolution with the "WiFi card not found" using a pi4b and 2.0.12b on the ground pi?

pilotnbr1, 2021-10-01:
qopenhd app issue-
The mAh/km reading start incrementing  even when the the drone is not  armed.

pilotnbr1, 2021-10-01:
> My apologies but just opened an issue to the same effect. Any resolution with the "WiFi card not found" using a pi4b and 2.0.12b on the ground pi?

No- havent looked into it yet. I dont have your exact card so a bit hard to troubleshoot but someone reported the same issue on a card I think I have so I will try that one and see whats going on

Initially we thought this was brought on due to an update.. but now that seems less likely

raphaelscholle, 2022-11-07:
Since we stopped development on 2.0x I'll close the issue here

closed 2022-11-07 by raphaelscholle