OpenHD/OpenHD #495: 2.0.12b WiFi card not detected on boot.

reichley, 2021-09-27:
Upon boot of ground pi (pi 4b) using the 2.0.12b image the screen shows "no WiFi card detected" and fails to boot. Using Alfa AC1200 AWUS036ACH (works fine under 2.0.8, same setup/connections/hardware)

[image]

pilotnbr1, 2021-09-27:
Thanks for this. Now we have a pattern since there are 2 of you guys

reichley, 2021-09-27:
You’re welcome! Thanks for all the hard work on the project! @pilotnbr1 @steveatinfincia and others…

MacDaddyFPV, 2021-09-27:
Can you please provide details about your setup, how it is wired and how you are powering it?

MacDaddyFPV, 2021-09-28:
Can you please try setting WIFI_HOTSPOT=N in your settings file and let us know the results?

paulick, 2021-09-28:
Same issues here. Tried WIFI_HOTSPOT=N. Nothing changed. Tried diffrent power-setups (usb-cable, supplies). Also the pi cm4 with no internal wifi shows the same error.

reichley, 2021-09-30:
As an update, also attempted to boot groundpi with 2.0.12b (same power supply) with both Alfa AWUS051NH and Alfa AWUS036NHA and receive same errors as photo above.

CkovMk, 2021-10-08:
Same issue. Ground Pi: Rpi4b + RTL8812AU card. I can confirm power is not the issue here.
By the way, It seems Air Pi with similiar setup boot up fine. Instead having a Pi4b, I use a CM4 for air pi tho.

MacDaddyFPV, 2021-10-11:
> @MacDaddyFPV GND: Pi 4GB v1.1, AIR: Pi CM4 Lite 2GB w/Wi-Fi, boot from TF card

There seems to be a bug introduced in the Kernel. Pease try adding the following to the bottom of config.txt as a workaround for now:
[all]
total_mem=3072

wx4cb, 2021-11-17:
do we know why this is needed? just curious because it obviously limits total memory to only 3G even on an 8 gig pi. just wondering what is causing it if it's known

raphaelscholle, 2022-03-24:
It's a kernel issue, which will be patched in 2.1. Pi's with more then 3GB ram currently have their USB-header broken, there is no easy fix to get rid on the problem in 2.0

raphaelscholle, 2023-01-31:
is fixed in 2.3

closed 2023-01-31 by raphaelscholle