OpenHD/OpenHD #497: ADSB function is not working in Open.HD 2.0.12B

VU2ATX, 2021-09-29:
Hi I have done detailed testing on RTL-SDR function of the latest release.
I have Pi3B+ as GroundPi, getting tested 4 RTL-SDR radios but getting "ADSB reply error" with all.
Then enabled debug and with "lsusb" command, I could see that each one of my RTL-SDR dongle is detected by the Pi3B+ when connected one by one.

Just for confirmation, I have also replaced the GroundPi with another Pi3B+ but same result, all RTL-SDR dongles get detected but give "ADSB reply error"

Have you tested this release at your end with any of the RTL-SDR dingle ? If yes, pls share a video clip if possible.
[image]
[image]
[image]

pilotnbr1, 2021-09-29:
Thanks for testing! 

There are substantial differences in the app between 2.0.8 and 2.0.12b with adsb. It was totally rewritten. However the base program which handles the receiving and processing of the signals from the sdr is the same - Dump1090-mutability

Dump1090-Mutability writes its results to a file which the qopenhd app reads. 

In the app if nothing is written to the dump1090 output file it will produce a "adsb reply error" this doesnt mean the sdr is not working. It could just be that no output is yet written to the file.. Be sure you are outside and that there are actually aircraft in your vicinity.
More info here https://github.com/OpenHD/dump1090-mutability/tree/openhd2.0
You may be able to look at the dump1090 log file or play with options to gain insight as to your particular issues

2.0.12b is confirmed working by one other user with sdr. I will also test this exact image with my sdr (nooelec dongle) but it was working for me when the last round of changes were made to adsb.

pilotnbr1, 2021-09-29:
Did you confirm if there was a difference between pi4 and pi3? Thought maybe I read something regarding this in the telegram group

pilotnbr1, 2021-09-29:
After looking at the could further- "adsb reply error" can only happen from the opensky side of adsb. This error has nothing to do with SDR.

I changed the ADSB error messages to be more exact in describing any errors. SO now they will say: "ADSB <sdr or opensky> <error here>"

VU2ATX, 2021-09-30:
> Did you confirm if there was a difference between pi4 and pi3? Thought maybe I read something regarding this in the telegram group
> 
> If this is the case perhaps you arent giving your pi3 enough power...

Hi, thank you so much for the reply. Yeah, last night I tested exactly same setup but with Pi4B (4GB) and there was no "ADSB Reply Error" alarm and after some time, I saw in the map an international flight was passing some 20km away my house in the village.

And it was indoor antenna.

VU2ATX, 2021-09-30:
> After looking at the code further- "adsb reply error" can only happen from the opensky side of adsb. This error has nothing to do with SDR.
> 
> I changed the ADSB error messages to be more exact in describing any errors. SO now they will say: ADSB sdr or opensky error here

I see, thank you so much, may I have the image to test please

VU2ATX, 2021-09-30:
> Thanks for testing!
> 
> There are substantial differences in the app between 2.0.8 and 2.0.12b with adsb. It was totally rewritten. However the base program which handles the receiving and processing of the signals from the sdr is the same - Dump1090-mutability
> 
> Dump1090-Mutability writes its results to a file which the qopenhd app reads.
> 
> In the app if nothing is written to the dump1090 output file it will produce a "adsb reply error" this doesnt mean the sdr is not working. It could just be that no output is yet written to the file.. Be sure you are outside and that there are actually aircraft in your vicinity. More info here https://github.com/OpenHD/dump1090-mutability/tree/openhd2.0 You may be able to look at the dump1090 log file or play with options to gain insight as to your particular issues
> 
> 2.0.12b is confirmed working by 1 (edit:now 3) other users with sdr. I will also test this exact image with my sdr (nooelec dongle) but it was working for me when the last round of changes were made to adsb.

I see, thank you so much for such a nice write-up. 
What is this "usb_claim_interface error -6" and "Failed to open rtlsdr device #0" errors messages I got (Pls refer my most recent screenshot in my main post.

pilotnbr1, 2021-09-30:
> I see, thank you so much, may I have the image to test please

There is no image built yet. 

The Qopenhd package gets re-built whenever we make commits. So there is the possibility for you to get new versions of the app before any type of official release. But it involves changing package sources to "testing" and then updating and upgrading the qopenhd package

pilotnbr1, 2021-09-30:
> I see, thank you so much for such a nice write-up. What is this "usb_claim_interface error -6" and "Failed to open rtlsdr device #0" errors messages I got (Pls refer my most recent screenshot in my [...]

I dont know without googling it. I am not familiar with those errors or the program that Daniel instructed you to try

pilotnbr1, 2021-09-30:
There is one more option if you want to try a new image. This is a very new automated build method for openhd. The "testing" source should have the latest changes- its all untested and was built on 30-9-21
https://github.com/OpenHD/Open.HD_Image_Builder/actions/runs/1291790429

VU2ATX, 2021-10-01:
> There is one more option if you want to try a new image. This is a very new automated build method for openhd. The "testing" source should have the latest changes- its all untested and was built on 30-9-21 https://github.com/OpenHD/Open.HD_Image_Builder/actions/runs/1291790429
> 
> EDIT: DONT try this.... It looks like its just building a 2.0.8 image. It needs updating

Hi Luke, Thank you so much for your continued support. It's okay then I shall wait for the builder script to be updated.

TTN-, 2021-12-26:
> I changed the ADSB error messages to be more exact in describing any errors. SO now they will say: ADSB sdr or opensky error here

So `ADSB Reply Error` error still suggests there are no aircraft within range?

pilotnbr1, 2021-12-26:
> > I changed the ADSB error messages to be more exact in describing any errors. SO now they will say: ADSB sdr or opensky error here
> 
> So `ADSB Reply Error` error still suggests there are no aircraft within range?

https://github.com/OpenHD/Open.HD/issues/497#issuecomment-930588547

pilotnbr1, 2021-12-26:
It means that the opensky network is not responding... Opensky is the interent source. This could happen because their website is down or the device you are using does not have an internet data connection

TTN-, 2021-12-26:
Brilliant, thank you pilotnbr1. I didn't realize I had to click carefully on the ADSB text, and then select SDR as the traffic source.
Wonder if in future it could be moved to the "APP" options screen, perhaps even both could be selected as a source just a thought but anyway, works now!!

Thank you for your hard work.

TTN-, 2021-12-26:
2.0.12B release notes mention:
`-Total rewrite of ADSB widget. Now agregates sources (David Contribution)`
Does this mean it already uses both sources if available? Or could the SDR be on the vehicle perhaps?

pilotnbr1, 2021-12-27:
@TTN- yes it agregates sdr groundpi, opensky internet, and any mavlink adsb messages that might be received by an on board adsb receiver (like a cube fc)

TTN-, 2021-12-27:
So SDR on vehicle is not supported right.

pilotnbr1, 2021-12-27:
No. Sdr on groundpi only.

closed 2023-01-31 by raphaelscholle