OpenHD/OpenHD #498: Inputs on RTL8811AU

rohitk1979, 2021-09-29:
Hi!
I have successfully been able to run OHD system on AR9271, RTL8812AU based cards.

However, for a micro build, these cards are coming out to be too bulky.

I have an RTL8811AU based card as well that is of perfect form factor for the micro build.

However, with this card, i am able to get the video transmission perfectly, but telemetry doesn't function for eg. with QGC.

Can you give me any pointers to what is different with this chipset so that i can look into modifying the code?

I couldn't find the datasheet for 8811 chipset anywhere on internet.

One more option could be to push the telemetry packets along with video packets.

How feasible that could be.  I am not too worried about RC as i will be using separate 2.4G RC system for control.

All i want is some Telemetry information along with the beautiful video.

pilotnbr1, 2021-09-29:
Its probably because some chipsets get different handling of the wifi frames. The frame handeling stuff is in wifibroadcast-base and a starting point would be rx.c tx.c 

It possible that 8811 is just not being recognised by ohd and isnt handles by the switch statement here
https://github.com/OpenHD/Open.HD/blob/8089a690b53454cb31a4bc2932f5f8c7648aa70b/wifibroadcast-base/tx_telemetry.c#L245

Pushing telemetry with video is a no go. There is no benefit to this and quite a few negatives.

rohitk1979, 2021-09-30:
Sure. Will look into the pointers and share feedback.

rohitk1979, 2021-10-06:
I tried to see the debug output of tx_telemetry.c

This is the case with RTL8812, wherein I am getting fully functional telemetry along with video:

[image]

Following its the screenshot with RTL8811, where i am getting only video - no telemetry or cpu usage etc.:

[image]

As you can see, in both cases, telemetry seems to be transmitted without any error.

What could be the bottleneck?
Where in the code should I see next for troubleshooting, rc_telemtry.c I believe?

rohitk1979, 2021-10-07:
I have observed that with 8811, neither telemetry functions, nor the shutdown command from ground to air pi works.

That implies the telemetry pipe in 8811 somehow isn't working. However, tx_telemetry debug above shows that there's no error.

closed 2023-01-31 by raphaelscholle