OpenHD/OpenHD #499: Feature request - Provission at ground Pi for offline maps so that every time internet access is not required for ADSB

VU2ATX, 2021-09-30:
@pilotnbr1 Hi Luke, every time ground Pi start, it is required to give internet access for downloading map and it becomes bit uneasy every time connecting the smartphone to the ground Pi for providing internet access via USB tethering.

Pls consider to give in future release, provission at ground Pi for offline maps so that every time internet access is not required for ADSB.

pilotnbr1, 2021-09-30:
This feature can be cross referenced here as well...

https://github.com/OpenHD/QOpenHD/issues/194

VU2ATX, 2021-10-01:
> This feature can be cross referenced here as well...
> 
> [OpenHD/QOpenHD#194](https://github.com/OpenHD/QOpenHD/issues/194)

Hi Luke, It's fine actually I was not aware of that.

Consti10, 2023-06-08:
closing, since it is a ground control application (QOpenHD) related task

closed 2023-06-08 by Consti10