OpenHD/OpenHD #500: MSP Telemetry Uplink

FellowWroclaw, 2021-10-04:
Inav and other MW based FC FW use MSP for datalink communication. Is it possible to make a transparent serial link via WBC? 
Also usb-serial bridge seems not to be working properly . It is working in system as ttyACM0, but no info nor in mavlink nor LTM modes . HardSerial mavlink is ok , no issues.

FC Omnibus F4V3
RPI3b+ both A&G
AR9271 A&G
RTL8812 A&G

wx4cb, 2021-10-04:
if you enable mavlink on the VCP port i'm not 100% that works in inav. I know if you enable it and MSP on a normal uart, then it's msp when disarmed, and mvlink/ltm when armed.

Would be nice to be able to use MSP while in the air, but i've yet to be able to inject mavlink directly into the ground pi from my 3DR radios etiher so... :D

FellowWroclaw, 2021-10-04:
> if you enable mavlink on the VCP port i'm not 100% that works in inav. I know if you enable it and MSP on a normal uart, then it's msp when disarmed, and mvlink/ltm when armed.
>
> Would be nice to be [...]

That's great. As for my tests mavlink on VCP works just fine on windows . but no way on open.hd. I've used telemetry after arm . but i need full nav control of the vehicle. Send WP's and execute commands

wx4cb, 2021-10-04:
if that's the case then you cant have mavinik on the usb side, as i've
stated before, when you arm, MSP will not become available and currently
afaik, the only way for inav to accept waypoints is via the MSP protocol.
but i could be wrong on that

FellowWroclaw, 2021-10-04:
I have nothing on usb recognized by open.hd. No LTM no Mavlink

wx4cb, 2021-10-04:
llike i said before, you would have to be armed for mavlink to come out of that port, now whether the usb port gets closed and re opened i don't know, but i personally would try a normal hardware uart, oh and btw, the device node should be /dev/ttyUSB0 or something like that.. ACM0 is either the onboard bluetooth or hardware serial.

FellowWroclaw, 2021-10-05:
> llike i said before, you would have to be armed for mavlink to come out of that port, now whether the usb port gets closed and re opened i don't know, but i personally would try a normal hardware [...]

Problem is there no spare uarts on FC
1- GPS 
3- I2C compass 
6-Serial RX Sbus 
VCP-MSP
That’s it , softserial is sending smth , but cannot be decoded ( MSP/LTM/Mavlink) baud 9600 thru 115200

wx4cb, 2021-10-05:
what flight controller is it ??? F411WSE ? btw, I2C doesnt apply here

FellowWroclaw, 2021-10-05:
FC Omnibus F4V3 . I2C in on UART3 pins. There are 3 uarts 1-3-6 and VCP

closed 2023-01-31 by raphaelscholle