OpenHD/OpenHD #518: not possible to get image to boot on Raspberry Pi Zero 2 W

mikaelsundin, 2021-11-06:
**Describe the bug**
Released images wont boot on Raspberry Pi Zero 2 W.

**To Reproduce**
1. Burn image from Open.HD-2.0.12b-buster.zip to sdcard on PC.
2. Move sdcard to Raspberry Pi Zero 2 W'
3. Power on rpi
4. HDMI: Screen is black is black, no raspberrys visibile. No activity led active.

**Expected behavior**
HDMI output and activivty led:s.

**Configuration and debug logs**

**Hardware:**
 - Air side
 -- Pi model - Raspberry Pi W 2
 -- Camera(s) 
 -- Wifi card(s)
 -- Other hardware
 - Ground side
 -- Pi model
 -- Wifi card(s)
 -- Ground Control Station(s)
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?

**Client devices (in case of hotspot/tethering issue)**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App - FPV_VR, QGC, and version
 - Version [e.g. 22]
 - Connection method

**Additional context**
Raspberry pi latest image boots as expected on the same RPI.
I suspect the Raspberry Pi Zero 2 W need a newer base raspberry pi lite image to work.

mikaelsundin, 2021-11-06:
Raspberry Pi Zero 2 W is close to Raspberry 3 in performance.

white-rabbitGR, 2021-11-06:
Raspberry Pi Zero 2 W may have similar performance to Raspberry Pi 3A+ but it's a completely different SBC. We are aware that it is not supported by OpenHD project.
Please have a look at the available supported SBC and choose one that suits best for you.

https://openhd.gitbook.io/open-hd/hardware/sbcs

Thank you for your input.

mikaelsundin, 2021-11-06:
Thx for quick response. I will check out the telegram chat. 
For me it is only important to have the Air Unit as small as possible. I will probably cut the PCB to get it smaller.

I think i can cut it down to 20x67mm.

closed 2021-11-06 by mikaelsundin