OpenHD/OpenHD #520: 2.1-Dev (2.0.0-404-g64a4d92) - AirPi - openhd_video - Failed to Create Pipeline With v4l2loopback Camera

Cyberunner23, 2021-11-12:
**Describe the bug**
In OpenHD 2.0.0-404-g64a4d92, when using a v4l2loopback based camera, the pipeline fails to be created. The code that autogenerates the pipeline includes a bitrate property to the v4l2h264enc element. The v4l2h264enc element has no such property. In fact, it seems it doesn't have any property for setting the bitrate. I would recommend using the x264enc element.

**openhd_video logs**
```
CameraMicroservice::setup()
Microservice::setup()
settings_path: /conf/openhd/ground
settings_file: /conf/openhd/ground/camera.conf
Configuring camera: 6
GStreamerStream::GStreamerStream()
GStreamerStream::setup()
Creating GStreamer pipeline
Setting up USB camera
Allowing gstreamer to choose UVC format
Pipeline: v4l2src name=picturectrl device=/dev/video0 ! video/x-raw ! videoconvert ! v4l2h264enc name=encodectrl bitrate=5000000 ! queue ! h264parse config-interval=-1 ! rtph264pay mtu=1024 ! tee name=t ! queue ! udpsink host=127.0.0.1 port=5620 t. ! queue ! udpsink host=127.0.0.1 port=5610
Failed to create pipeline: no property "bitrate" in element "encodectrl"
GStreamerStream::start()
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use a v4l2loopback based camera, or a USB camera (AFAICT this happens with any USB camera)
2. Check the logs for openhd_video
3. Notice the error message `Failed to create pipeline: no property "bitrate" in element "encodectrl"`

**Expected behavior**
The pipeline is successfully created

**Configuration and debug logs**
Default configs

**Hardware:**
 - Air side
 -- Pi model: RaspberryPi 4B
 -- Camera(s): OAK-D with software to push the RGB cam through a v4l2loopback device
 -- Wifi card(s): rtl8814AU based wifi device
 -- Other hardware: None
 - Ground side
 -- Pi model: RaspberyPi 4B
 -- Wifi card(s): rtl8814AU based wifi device

**Additional context**
We are using an [OAK-D](https://shop.luxonis.com/products/1098obcenclosure) camera. This camera has built in ML hardware and can only be used with an SDK, it does not act like a regular camera. We are using it to do some depth perception on the drone side for collision avoidance. We also want to be able to send the RGB camera data to the ground pi. To do so we have a script that sets up the camera with the depth perception as well as sets up a GStreamer Pipeline with an appsink pushing frames into a v4l2loopback device.

pilotnbr1, 2022-02-03:
@careyer @MacDaddyFPV @patemil @MarioFPVdev @raphaelscholle @cq112358  is this resolved?

MacDaddyFPV, 2022-03-24:
It seems like this has been updated in the code but I do not have a usb camera to test this.
https://github.com/OpenHD/Open.HD/blob/e255c90bd7f6cd2f48df129e34df9dabbb628e6f/openhd-video/src/gstreamerstream.cpp#L443
@Cyberunner23 Are you able to test the latest build and provide feedback as to whether it is working?

Consti10, 2023-02-01:
can be closed, please try again with 2.3.x - evo and report.

closed 2023-02-01 by Consti10