OpenHD/OpenHD #526: The serial communication speed is 115200, which is too slow. Multiple UAVs cannot be online. Is there a way to set it? thank you

wqq3000, 2021-11-23:
The serial communication speed is 115200, which is too slow. Multiple UAVs cannot be online. Is there a way to set it? thank you

wx4cb, 2021-11-23:
Lol... i would never thought i would see the day when someone said 115k2 is too slow..

When u say multiple uav cannot be online at the same time, online in what sense? Isnt open hd a 1-1 relationship air to ground? Also, dont forget, in standard serial communications the higher the baud rate, the  more errors can be introduced at greater distances. 

I think more detail outside of  "its too slow" is needed imho

wx4cb, 2021-11-23:
Also, its potentially a hardware limitation in the uart as well

CkovMk, 2021-11-24:
Probably you could try high speed USB-UART dongles, such as FT232H based dongles.

wqq3000, 2021-11-24:
hi!wx4cb,
   I'm trying to fly in formation with multiple drones. Multiple flight controllers communicate with one OpenHD board. 115200 can hardly communicate with 3-4 UAVs. It's too slow. Raspberry pie OS serial communication can be adjusted to 921600
thank you!

wqq3000, 2021-11-24:
> 嗨！wx4cb，
> 我正在尝试使用多架无人机编队飞行。多个飞行控制器与一个 OpenHD 板进行通信。115200很难与3-4架无人机通信。太慢了。树莓派OS串口通讯可以调到921600，openhd不行！
> 谢谢你！

wx4cb, 2021-11-24:
@wqq3000  yiou need to find a faster link, there is no way a serial link even at 921600 is going to be able to keep up with that amount of data. the overheads are just too much. and the software processing is probably going to be too slow as well. you would be better off using a network based link

wqq3000, 2021-11-26:
> @wqq3000 你需要找到一个更快的链接，即使是 921600 的串行链接也无法跟上如此大的数据量。管理费用太多了。并且软件处理也可能会太慢。你最好使用基于网络的链接

Hello! If the communication between flight control and OpenHD is higher than 115.2k, the ground station will lack data. We want to increase the communication rate a little, but we don't know how to use it. thank you

raphaelscholle, 2023-01-31:
in 2.3 higher data rates are possible

closed 2023-01-31 by raphaelscholle