OpenHD/OpenHD #538: 2.0.1b Split Video

akurekin, 2021-12-05:
Hi, I am testing a new 2.0.12b version on my setup: AirPi - raspberry Pi Zero + VEYE mipi 327 camera; Ground PI - raspberry pi 4B connected over HDMI. The screen is split into two equal parts and the top part appears below the lower one. Any suggestions to resolve this problem?

Thanks

akurekin, 2021-12-05:
I would like to clarify that only the camera video image is split into two parts but OSD is fine. So, it is not an issue with HDMI, it's an issue with video.

MacDaddyFPV, 2021-12-10:
I have seen this previously with an incompatible resolution for the camera vs display. What are you using as a display? What are your camera settings? Have you set them on both the ground and airpi?

akurekin, 2021-12-11:
Hi, here is what I did. Firstly I tried the default settings and the camera image was split in two parts. By default I have WIDTH=1280 HEIGHT=720 in openhd-settings-1.txt.
Then I tried 1920x1080 and have got just a black screen after booting up. 
Then I tried setting hdmi_group=1 hdmi_mode=16 in the config.txt with not effect, just a black screen. Also tried framebuffer_width=1920 framebuffer_height=1080 and all the same no effect. Connected to my TV over HDMI cable, the TV was showing resolution 1920x1080. I'll very appreciate your help!

Thanks

akurekin, 2021-12-13:
I made some progress with my question. It looks like Pi Zero Air is not capable of running at 1920x1080 resolution, the one that OpenHD 2.0.12 + Veye are running on. That was the first problem. Now the second one. It looks like the parameters at Ground Pi cannot be changed just by editing the  openhd-settings-1.txt , only through the App. Is this correct?

closed 2023-01-31 by raphaelscholle