OpenHD/OpenHD #545: UDP video stream on qgoundcontrol-openhd version not working

charlesmarseille, 2022-01-02:
**Describe the bug**
As title says. No image whatsoever. Removing the app and installing the standard (non openhd version, directly from play store) works, got solid feed on 2 devices at the same time.

**To Reproduce**
3a+ for airpi with ac56
3b+ for groundpi with awus1900
qgroundcontrol+openhd version taken directly from the wiki (https://drive.google.com/open?id=1sHjUKq0GJLkRrzAMFJs83avuvvMcVyQd)

Install, open, set udp and 5600 port. Check if image is active.

**Expected behavior**
Live feed should be obtained directly.

**Hardware:**
 - Air side
 -- Pi model : 3a+
 -- Camera(s) : P5V04A
 -- Wifi card(s) : ac56
 - Ground side
 -- Pi model : 3b+
 -- Wifi card(s) : awus 1900
 -- Ground Control Station(s) : QGroundControl+openhd version
 -- Hotspots : samsung tablet and samsung phone connected to AP (internal wifi chip from 3b+)
 - RC/telemetry uplink
 -- None

pilotnbr1, 2022-01-03:
I dont think anyone has used that app for years.. we should probably just take it down as an option...

Consti10, 2023-02-04:
install normal QGroundControll and use 2.3.1-evo

closed 2023-02-04 by Consti10