OpenHD/OpenHD #552: 2.1 Dev - Innomaker IMX290 not recognized

careyer, 2022-01-17:
for 2.1 Dev Innomaker IMX290 cameras do not get recognized.

- no detected camera shows up in` /tmp/manifest`
- accordingly unit starts up as GrounbPi
- tried also with `dtparam=i2c_vc=on ` enabled in config.txt

Tested with:
- Pi3B+/Pi4B
- Innomaker IMX290

Hope this can be fixed! Awesome camera! ;-) - Cheers!

careyer, 2022-02-18:
Some good aditional information can be found here:

https://www.inno-maker.com/product/mipi-cam-290/

http://wiki.veye.cc/index.php/VEYE-MIPI-290/327_for_Raspberry_Pi

http://wiki.veye.cc/index.php/VEYE_CS_Camera_for_Jetson_TX2

MacDaddyFPV, 2022-03-24:
VEYE camera detection is implemented and has been tested on the VEYE IMX462 on a RPI 3b+.

The pipeline to run the VEYE cameras with the new pipeline setup is complex as it requires to run custom code from VEYE.

closed 2022-10-17 by raphaelscholle