OpenHD/OpenHD #553: 2.1 Dev - AirJetson not recognized by GndPi

careyer, 2022-01-17:
I am trying to get a hybrid setup working between AirJetson (refered to as: Jetson,air) and GndPi (Pi,gnd)

- Pi,air(2.1Dev) --> Pi,gnd(2.1Dev) Setup works
- Jetson,air(2.1Dev) --> Pi,gnd(2.1Dev) Setup does not work

When Jeston,air(2.1Dev) is booted up during a perfectly fine working "Pi,air(2.1Dev) --> Pi,gnd(2.1Dev)" session,  it will heavily interfere with the working setup and the video reception breaks up completely (most presumeable due to packet collision), 

Verdict: The Jetson,air is in fact transmitting but for some reason does not get recognized by Pi,gnd as a valid air unit.

pilotnbr1, 2022-02-03:
@careyer is this more of an atheros driver issue?

careyer, 2022-02-03:
Yes... this is an Atheros issue. Without the htc_9271.fw patch the airUnit's injection speed is so low (<180-200KiB/s) that the gndStation can not even make sense of the data it receives. Hence telemtery and video data get completely corrupted. Atheros fix needed to solve this problem :-)

pilotnbr1, 2022-02-03:
https://github.com/OpenHD/Open.HD/issues/573 covers this