OpenHD/OpenHD #577: 2.1 Dev - Support for Thermal cameras incomplete (Flir driver)

careyer, 2022-02-04:
I sampled manifest and config files for the following cameras:
- hti-301
- FlirOne (2nd Gen)

here are the results:

1. **hti-301 does get recognized**

/conf/openhd/vehicle1/camera.conf
```
[Camera]

# DO NOT change these, they are used for identifying this camera to the system
type=uvc
bus=usb-70090000.xusb-2.3
name=T3
vendor=Infiray
# end identification section
```
/tmp/manifest
```
{
    "camera": [
        {
            "bus": "usb-70090000.xusb-2.3",
            "endpoints": [
                {
                    "device_node": "/dev/video1",
                    "formats": [
                        "YUYV 4:2:2|384x292@25"
                    ],
                    "support_h264": false,
                    "support_h265": false,
                    "support_mjpeg": false,
                    "support_raw": true
                }
            ],
            "index": 0,
            "name": "T3",
            "pid": "0001",
            "type": "uvc",
            "vendor": "Infiray",
            "vid": "1514"
        },
        ...
}
```

2. **FlirOne DOES NOT get recognized**.
I suspect that in order to be regonized the Flir Driver must be loaded first. While hti-301 has the behavior of a normal USB-webcam the Flir camera stream outputs an arbitrary datastream. In this stream two logical video streams are transported (one for the visual / on for the thermal view). The driver then de-muxes those streams and binds them to two v4l2loopback devices.
Maybe the Flir driver has yet not been compiled for jetson?

Best,
careyer

careyer, 2022-02-06:
Turns out Jetson does not have the v4l2loopback kernel module installed
[image]

Instructions on how to build it for the kernel we use and install it can be found here:
https://github.com/umlaeute/v4l2loopback
(see section: https://github.com/umlaeute/v4l2loopback#dependencies)

raphaelscholle, 2023-04-15:
We currently only support Thermal cameras which act as uvc camera, for all these cameras a loopback is not needed, the kernel does include one, but till the dev team has a camera which needs an loopback, we can't really develop there.

Also the Jeson will be discontinued by openhd in the next release.

closed 2023-04-15 by raphaelscholle