OpenHD/OpenHD #584: "32-bit" deb packages are built on a 64-bit Docker, resulting in 64-bit binaries

MorrisR2, 2022-02-12:
In Dockerfile-armhf-raspbian-bullseye, it uses ghcr.io/openhd/bullseye-lite:v2.2

That results in arm64 binaries in a package that claims to be armhf.
It needs to be a armhf (32 bit) Docker image to build the armhf packages.

Also note 64-bit is detrimental if you aren't running 8GB of RAM, so there is good reason to keep the package armhf.

pilotnbr1, 2022-02-14:
If you want to get directly involved we can always use the help!

MorrisR2, 2022-02-14:
I likely will get more involved, though maybe not by building this Docker image.  I'm a 25-year Linux programmer and sysadmin, so I'll probably do some things.

What might be helpful would be an architecture document that shows how the parts are supposed to fit together. I see there are several micro services. For example, I noticed when you boot up there is a socat instance running *in the foreground*, blocking the terminal. It's not obvious what that's for or why it's running in the foreground instead of as a service.

pilotnbr1, 2022-02-14:
@MorrisR2 having more pros involved would be great. I think everyone directly involved with 2.1 (master branch of openhd) at this point is a hobbyist programmer (myself included) - some are very capable though! As you may or may not know, Stephen was working on 2.1 and left about a year ago for personal reason. He left 2.1 at about 90 percent complete. There has been quite a bit of progress made in the last few months though. We have working images on pi and jetson doing h265 with low latency on the new refactored code (all the management bash was ripped out and replaced with C++ and boost library). If you want to be more involved/in the know, there is a very active telegram group , maybe too active some days :p  . If interested in telegram group let me know

So there really isnt an architecture doc- there is a diagram showing all the udp/tcp connections and ports. Otherwise I can direct you to the current Project/issues  https://github.com/OpenHD/Open.HD/projects/5  and this issue which outlines things a bit https://github.com/OpenHD/Open.HD/issues/402 . We need some better documentation for sure- most of the design decisions have already been made. Only remaining decision on basic design are some questions regarding handling of settings between qopenhd (gui controlling) and air/ground sbc

Regarding Socat.. The project is full of quirky, hacky things. Some are necessary ,others are left over from someone else. Things got quite frankesteined in 2.0 (2.1 , master branch of openhd, cleans it up alot ). Specifically I think socat is used when piping stdin to udp and splitting on 1024 bytes (something like raspivid -t 0 -o - | socat -b 1024 GOPEN:/dev/stdin UDP4-SENDTO:127.0.0.1:5600) .. But that doesnt really explain why its in the foreground to me... See we need help haha

closed 2022-02-18 by MarioFPVdev

MorrisR2, 2022-02-19:
Thank you very much for your informative response @pilotnbr1 
I can confirm I get 32-bit binaries after running apt update - though the image I built today got 64-bit when it was built.

I'll probably catch you guys on the Telegram and do more work on OpenHD after I finish my thesis in about 8 weeks.
(I'm fooling myself here, I'll probably work on OpenHD *instead of* my thesis, which is really dumb - but more fun.)