OpenHD/OpenHD #617: No Hotspot Active on raspberry pi 4 2GB RC 2.0.12

AbrahamB9, 2022-03-11:
**Describe the bug**
I have been testing OpenHD RC 2.0.12  on my Raspberry pi 4 2GB of Ram and it´s working properly just the Wifi Hotspot can´t be enable, on the booting messages says that there´s no hardware available for Hotspot. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
OpenHD booting properly but no Wifi Hotspot available

**Configuration and debug logs**
Wifi Hotspot on Auto and also tried ON settings.

**Hardware:**
 - Air side
 -- Raspberry pi Zero W
 -- V1 camera
 -- 

 - Ground side
 -- Raspberry pi 4 2GB 
 -- 
 -- QOpenHD app
 -- Hotspots - Wifi
 - Mavlink telemetry
 -- Rc over TBS crossfire

**Client devices (in case of hotspot/tethering issue)**
 - Device: Android phone

zhuwei, 2023-01-06:
I tried to recompile the firmware, unfortunately failed

raphaelscholle, 2023-01-31:
should be resolved in 2.3

closed 2023-01-31 by raphaelscholle