OpenHD/OpenHD #620: 2.1 - dev - Builder state

pilotnbr1, 2022-03-23:
With all the recent changes and all the platforms there are builder bugs and uncertainty what is broken and how.
Assume 2.1-testing as the package source for everything.
The goal is to have a stable build on all platforms so that a release can be tagged and thereby have stable build packages available. Possibly should rename package repo "2.1" 

TESTED WORKING BUILDS

-  Jetson Air 2GB (2GB image) --> Jetson Ground 4GB (4GB image)

BROKEN

- Bullseye Pi 
- Buster Pi

pilotnbr1, 2022-03-23:
BUSTER PI ERRORS
Suspect DRM-issues with bullseye

With with the line "dtoverlay=vc4-kms-v3d" in config.txt we see these 2 errors
-openhd kernel: raspberrypi-cpufreq: probe of raspberrypi-cpufreq failed with error -2
-openhd QOpenHD[537]: Could not open DRM device /dev/dri/by-path/platform-soc:gpu-card (No such file or directory)

pilotnbr1, 2022-03-23:
BUSTER PI
not booting

reedev, 2022-03-23:
newer rpi4 cannot boot buster image

MacDaddyFPV, 2022-03-24:
> BULLSEYE PI ERRORS Suspect DRM-issues with bullseye
> 
> With with the line "dtoverlay=vc4-kms-v3d" in config.txt we see these errors (these might be out of order)
> 
> -openhd kernel: raspberrypi-cpufreq: probe of raspberrypi-cpufreq failed with error -2
> 
> -openhd QOpenHD[537]: ERROR: ld.so: object '/opt/vc/lib/libGLESv2.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
> 
> -openhd QOpenHD[537]: ERROR: ld.so: object '/opt/vc/lib/libEGL.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
> 
> -openhd QOpenHD[537]: Could not open DRM device /dev/dri/by-path/platform-soc:gpu-card (No such file or directory)

When testing on Bullseye QOpenHD closes with "error while loading shared libraries: libicui18n.so.63"
This is a debian 10 package and should be using libicu67. https://debian.pkgs.org/11/debian-main-arm64/libicu67_67.1-7_arm64.deb.html
I suspect that the way Openhd-qt was built for Bullseye (using the buster package?) is causing these issues.

reedev, 2022-03-24:
> 

note: Jetson 2GB --> Jetson 4GB is working ok when network cable is not connected

reedev, 2022-03-24:
Even when I disable the qopenhd service, qopenhd is started at next boot. What's happening there?

raphaelscholle, 2022-03-30:
Bullseye investigation:

Jan 28 01:23:26 openhd QOpenHD[421]: ERROR: ld.so: object '/opt/vc/lib/libGLESv2.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Jan 28 01:23:26 openhd QOpenHD[421]: ERROR: ld.so: object '/opt/vc/lib/libEGL.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Jan 28 01:23:26 openhd systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Jan 28 01:23:26 openhd systemd[1]: Finished Update UTMP about System Runlevel Changes.
Jan 28 01:23:26 openhd systemd[1]: Startup finished in 984ms (kernel) + 35.507s (userspace) = 36.492s.
Jan 28 01:23:26 openhd QOpenHD[421]: /usr/local/bin/QOpenHD: error while loading shared libraries: libicui18n.so.63: cannot open shared object file: No such file or directory
Jan 28 01:23:26 openhd systemd[1]: qopenhd.service: Main process exited, code=exited, status=127/n/a
Jan 28 01:23:26 openhd systemd[1]: qopenhd.service: Failed with result 'exit-code'.

first step to get it working looks like to get libicu18n working

raphaelscholle, 2022-03-30:
with the hacked libicu (which really needs to be rebuild)
we get

Mar 30 20:03:47 openhd QOpenHD[434]: ERROR: ld.so: object '/opt/vc/lib/libGLESv2.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Mar 30 20:03:47 openhd QOpenHD[434]: ERROR: ld.so: object '/opt/vc/lib/libEGL.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Mar 30 20:03:47 openhd systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Mar 30 20:03:47 openhd systemd[1]: Finished Update UTMP about System Runlevel Changes.
Mar 30 20:03:47 openhd systemd[1]: Startup finished in 921ms (kernel) + 36.390s (userspace) = 37.312s.
Mar 30 20:03:48 openhd QOpenHD[434]: LocalMessage::LocalMessage()
Mar 30 20:03:48 openhd QOpenHD[434]: LocalMessage::init()
Mar 30 20:03:48 openhd QOpenHD[434]: ManageSettings::ManageSettings()
Mar 30 20:03:48 openhd QOpenHD[434]: Loading settings file
Mar 30 20:03:48 openhd QOpenHD[434]: Could not open DRM device /dev/dri/by-path/platform-soc:gpu-card (No such file or directory)
Mar 30 20:03:48 openhd QOpenHD[434]: Could not open DRM device
Mar 30 20:03:48 openhd systemd[1]: qopenhd.service: Main process exited, code=killed, status=6/ABRT
Mar 30 20:03:48 openhd systemd[1]: qopenhd.service: Failed with result 'signal'.

raphaelscholle, 2022-03-30:
My current idea is to try if there is an issue with our qt-raspberry build, I think I'll add crosscompiling for it, another thing where can remove drone.io building and go to github runners

pilotnbr1, 2022-04-30:
This issue is not being updated with progress changes and is superceded by the goal of an alpha image. I am closing it to get rid of clutter. If someone disagrees then they can just reopen it

closed 2022-04-30 by pilotnbr1