OpenHD/OpenHD #621: 2.1 Dev - Enable Libcamera compatibility on Bullseye

MacDaddyFPV, 2022-03-24:
There are a number of cameras which could be made compatible with the use of Libcamera on Bullseye.
This will need a separate detection method along with a separate pipeline to use libcamerasrc.  

On a Bullseye build this should detect and setup any Libcamera compatible camera.
It should then start a pipeline using the detected camera settings.

This change will be only available on Pi unless other systems support Libcamera in the future.

MacDaddyFPV, 2022-03-24:
Initial testing shows that libcamera is not recognising cameras. This will need to be fixed before any camera detection is put in place.

raphaelscholle, 2022-10-17:
libcamera is working in 2.2

PR #669 (2022-08-17): Libcamera
Fixes #621 
`libcamerasrc` and `v4l2h264enc` used in gstreamer pipeline.
`libcamera` used to detect cameras.
closed 2022-10-17 by raphaelscholle