OpenHD/OpenHD #623: OpenHD video player module / service.

Consti10, 2022-03-31:
Hello,

Let's write the specifications for a generic OpenHD video playback module here. Once they are accepted, we do not want to change them - so we need to take our time.

**Name:**
 OpenHD video player

**What does it encapsulate:**
Decode the live video stream on the ground station and display it.

**Interfaces:**
Input: encoded video data, e.g. h264, h265 and MJPEG.
Output: Video to the screen, latency and other statistics.

**Dependencies:** 
QopenHD, but we can seperate these two
If seperated, following issues/ limitations arrise:
1) QOpenHD OSD needs to be transparent.
2) It won't be possible to resize it via QOpenHD UI (at least not easily feasible)
3) More than one video stream cannot be easily displayed - swapped between.

**How to implement:**
Let's start with the input interface and the specifications for the input data stream:
**Input interface:**
H264, H265 or MJPEG
Should already be depacetized regarding FEC and so forth
For H264/ H265, the _requirements_ are:
A valid SPS/PPS needs to be fed at some point, until that happens the decoder cannot start decoding.
It needs to handle changing resolution(s) on the fly. (see output interface).

**Output Interface:**
Screen (of course). Statistics via another callback (decoding latency - estimate or exact, current width, height)
Regarding changing resolution(s)/ frame-rates:
This module shall be able to just handle a changing resolution of the input video stream. Since the video output is stretched to the full screen anyways  (adhering to the video ratio of course) I don't see any problems here.

**Platform dependency:**
Yes, there are (big) platform dependencies. RPI offers hello_video for h264, ffmpeg for h264 and h265.
Jetson offers the nvidia sdk, but mostly gstreamer for h264 and h265 playback.
MJPEG is nieche, I don't know exactly how things are there - perhaps gstreamer or ffmpeg is usefully for that.

Remarks:
I'd prefer to implement the input interface in simple c++ fashion, via callbacks. If code dependencies are an issue, we can use UDP (for example) as a bridge, but for low latency this needs to be avoided. I'l upload a draft soon.

This c++-implementation can be platform independent (exactly this dependency should be hidden by this module).

Question: Does this already exist in OpenHD somewhere:
No, we have weird pipes, platform dependencies everywhere. 

Question: How big is the time investment: I cannot say, I aim for 1 week after the interface(s) are drafted

Question: Is this module testable: Yes, really well. We can just write a dummy rx stream that follows the input specifications.
(Ideas: Test changing resolution/framerate, test h264, h265 mjpeg).

Consti10, 2022-03-31:
Dummy for code:
https://github.com/OpenHD/XOpenHDVideo

Consti10, 2022-03-31:
@Petre
that looks good. doesn't OpenMAX provide a HAL for h264/265 decode? it's definitely available for raspberry pi and Jetson

MarioFPVdev, 2022-03-31:
Could we use a Qt ffmpeg player inside QOpenHD and then use an .sdp file to stream?

Consti10, 2022-03-31:
@pilotnbr1 what do you think, especially in regard to QOpenHD ?

Consti10, 2022-03-31:
> Could we use a Qt ffmpeg player inside QOpenHD and then use an .sdp file to stream?

Well, kinda, but that won't work on all platforms automatically. Also makes it impossible to do any decoding optimizations.

Consti10, 2022-03-31:
@pilotnbr1 
"With qopenhd the biggest stumbling block might be that transparency seems to be handled differently by different platforms.

I only tested/used transparent background on x86 linux which worked well. I think i did android too without a problem

[@Mario_FPV]() tried transparency on the pi (maybe jetson too) without success. Maybe he can comment on that

No data on ios or windows regarding transparent background...

Otherwise I like all the ideas. I do wonder how pip might be handled within qopenhd with this... but crawl walk run"

Consti10, 2022-03-31:
@pilotnbr1 
With qopenhd the biggest stumbling block might be that transparency seems to be handled differently by different platforms.

I only tested/used transparent background on x86 linux which worked well. I think i did android too without a problem

[@Mario_FPV]() tried transparency on the pi (maybe jetson too) without success. Maybe he can comment on that

No data on ios or windows regarding transparent background...

Otherwise I like all the ideas. I do wonder how pip might be handled within qopenhd with this... but crawl walk run

Consti10, 2022-03-31:
> @pilotnbr1 With qopenhd the biggest stumbling block might be that transparency seems to be handled differently by different platforms.
That's a bummer. In theory, it should work. QOpenHD just needs to render into a HW composer surface with transparency.

> 
> I only tested/used transparent background on x86 linux which worked well. I think i did android too without a problem

I did not consider android/ apple in this regard, since there things are completely different anyways (video is forwarded by the ground pi, apple / android only allow their specific decoding APIs that are well documented).

> 
> @Mario_FPV tried transparency on the pi (maybe jetson too) without success. Maybe he can comment on that
> 
> No data on ios or windows regarding transparent background...
> 
> Otherwise I like all the ideas. I do wonder how pip might be handled within qopenhd with this... but crawl walk run

closed 2022-11-07 by raphaelscholle