OpenHD/OpenHD #625: 2.1 Dev - Enable VEYE camera compatibility

MacDaddyFPV, 2022-04-02:
The methods used in 2.0 for VEYE cameras will no longer work with 2.1.

The cameras will need to be detected and then a Gstreamer pipeline setup for the camera.

MacDaddyFPV, 2022-04-02:
Camera detection has been setup and camera is being detected on pi3b+ bench test.
Camera is not being detected on pi4b due to an i2cdetect issue.

MacDaddyFPV, 2022-04-02:
As the VEYE cameras use custom software to run the camera, it is not able to be used in the standard gstlaunch pipeline. V4L2 driver method is being investigated.

MacDaddyFPV, 2022-04-02:
V4L2 drivers are not available from VEYE for the 4.19 Kernel so this will not work on the current Buster version. Investigating if it will work for Bullseye.

MacDaddyFPV, 2022-04-02:
Bullseye is using the 5.10.0 Kernel which there are still no drivers available from VEYE. We would need to upgrade the Kernel to one that has a driver for this method. See https://github.com/veyeimaging/raspberrypi_v4l2/tree/main/release/driver_bin

raphaelscholle, 2022-04-27:
Kernel updated ;)

raphaelscholle, 2022-10-17:
veye is working in 2.2 on pi and jetson

closed 2022-10-17 by raphaelscholle