OpenHD/OpenHD #626: QOpenHD - define interactions between OpenHD and QOpenHD

Consti10, 2022-04-06:
This issue follows up on https://github.com/OpenHD/Open.HD/issues/623

First, let me put it out there - Having the video rendering done by qt / QOpenHD reduces complexity a lot. The reason I was favouring an external video renderer (not in QOpenHD) before was mostly because I was worried about latency and not being able to do the OSD and video composing by the HW composer.

However, after working a bit with qt and especially after looking through the moonlight source code I actually think it is possible to do exactly the above in qt / QOpenhd.

Therefore, it makes more sense to define the interface for the interaction(s) between OpenHD and QOpenHD together regarding video.

Draft:

**QOpenHD**

Functionalities:
1) Render the Video and display it to the User (done/todo)
2) Render the OSD and display it to the User (done already)
3) Change Settings for OpenHD, (todo, currently a slugfest) -> see telemetry part
5) Handle RC via joystics if wanted by the user (re-use)

Anything missing here ?

Now let's see what is the simplest way for QOpenHD to interface with OpenHD/ the ground pi to implement the above functionalities.

1) Not having any code dependencies between QOpenHD and OpenHD is desired. This means we cannot pass data via c/c++ interfaces, but need a platform-provided way of doing so. 
UDP is doing this at the time, and while not ideal (latency and copying of data) we can keep that for now.

2) With the decision to do everything related settings, telemetry via mavlink we can get rid of multiple links between QOpenHD and OpenHD that existed in the previos implementation(s) - like the custom raw data packets for the wifibroadcast stats.

This makes the required communication between QOpenHD and OpenHD to implement the above functionalities quite simple -

1) Unidirectional link for raw video 
OpenHD (ground pi) -> | raw h264/h265 encapsulated in rtp, via UDP | -> QOpenHD

2) Bidirectional link for telemetry
OpenHD (ground pi) <-> | custom mavlink via TCP | <-> QOpenHD 

Only these 2 links are quite complex now -
regarding 1) 
1.1) A valid SPS/PPS needs to be fed at some point, until that happens the decoder cannot start decoding.
1.2) The resolution/framerate/format might change, QOpenHD needs to automatically react to that

regarding 2)
2.1) we need custom mavlink messages, mostly for 
_wifibroadcast_ 
_telemetry_
_settings_
_log messages_
2.2) it adds a dependency between QOpenHD and OpenHD - our custom mavlink messages. We have to duplicate the mavlink library to OpenHD and QOpenHD - but I think this is still the simplest and best solution.

Consti10, 2022-04-06:
Testing:
Regarding video: Start a dummy encoded video stream with gstreamer:
gst-launch-1.0 -v ximagesrc ! video/x-raw,framerate=20/1 ! videoscale ! videoconvert ! x264enc tune=zerolatency bitrate=500 speed-preset=superfast ! rtph264pay ! udpsink host=127.0.0.1 port=5600

MarioFPVdev, 2022-04-13:
I think we should use the port 5620/5621(PiP) for video since the router is archived.

Consti10, 2023-01-28:
Finished, in short:
rtp udp for primary video on 5600
rtp udp for secondary video on 5601
mavlink uses the same approache(s) like QGroundControl by default.

closed 2023-01-28 by Consti10