OpenHD/OpenHD #627: OpenHD debugging script - ideas and demands

raphaelscholle, 2022-04-07:
Since many nightlies are currently broken and don't get tested before being uploaded,we had the idea to create a debugging script.
This script should be executed on boot and handle all the testing, we need to verify a working image.

To get started on building those script(s) we need to collect a few suggestions and ideas.

**What do we need to be tested**

-Is OpenHD installed correctly
-Are all Services running
-Are monitor-mode drivers installed
-Are wifi-cards put into monitor state
-Are packages send
-Is Qopenhd running
-Is the camera detected
...

**How can we test this**
- check for files with if [ -z "$(ls -A .)" ];
- export systemctrl to file
-
- export iwconfig to file
-
- 

**Where do we post the issues, what file format,...**
- we could add a file to /config/boot (like raspbian does it with ssh), which is checked while booting, and executes a crontab 
- We could write files to /config/boot/debug , that space should be writable, is accessible from any computer when plugging in the sd-card (we have about 190MB space)

Consti10, 2022-04-07:
To me, it sounds as if some if it could be done by proper tests for the specific OpenHD modules itself. 

What is a bit related to this - stephen started writing a OpenHD logger - that is, instead of cout<<"blabla" we should use OHDL<<"blabla" and this logger then stores the logs in either a text file or forwards them to QOpenHD for output.

raphaelscholle, 2022-04-07:
The idea behind this custom script is, that it runs without OpenHD running, it should be much lower "level" to debug even if QopenHD,OpenHD, .. is failing

MacDaddyFPV, 2022-04-08:
Maybe this should also export the /tmp/manifest to see what has been detected.

closed 2023-01-31 by raphaelscholle