OpenHD/OpenHD #628: Milestone 1 Alpha 1

Consti10, 2022-04-07:
This is a milestone for all OpenHD developers - to motivate and bring us closer together.

For the first milestone,let's go back to the roots and go for the most simplest functionalities. That is
a) having a video feed and 
b) mavlink based telemetry (downlink only).

While these functionalities have been already demonstrated multiple times by developer-created releases, they were always created by individual persons and never verified by other developers involved. Above functionalities are a simple and achievable goal that can work as a baseline for future collaboration. 

We also do not need to remove any features that exceed the above specifications - we just need to make sure that we test the above on all platforms (raspberry pi, jetson,pc(x86) ).

How shall we define this milestone as done ?
1) We have working images that fulfill the above specifications
2) We have a release or some other means of downloading these images

This milestone was created by @careyer, @raphaelscholle and @Consti10. Further communication with @pilotnbr1 is required. 

All hands on ! Let's be done in 1 week.

raphaelscholle, 2022-04-10:
Issues currently working on ;   

- [ ] Image Builder
        - [X] Create clean milestone repository
        - [X] Remove Nightlies
        - [X] Create new milestone branch 
        - [X] Include debug-script (WIP)
        - [X] Create new workflow-file, which includes all builds (and maximize buildspace)
        - [] Include Qopenhd
        - [] Include current kernels

MacDaddyFPV, 2022-04-11:
Milestone image boots on Pi4 but will not boot on Pi3b+. Replacing the Kernel with one from a previous image allows it to boot so I suspect an issue in the Kernel.

MacDaddyFPV, 2022-04-11:
I looked into this further. I was able to get it to boot by replacing the bcm2710-rpi-3-b-plus.dtb file from a standard rpi os lite image so I suspect we are doing something strange with this file. I will dig further.

raphaelscholle, 2022-04-11:
Kernel is currently bild from 2.1-multi branch, thanks for the input, I'll look into that later

Consti10, 2022-04-11:
How far away are we from a bootable image for rpi (bullseye) and jetson ?

raphaelscholle, 2022-04-11:
I'm waiting for QopenHD, after that it shouldn't be much work left

Consti10, 2022-04-11:
What do you wait for regarding QOpenHD ?

raphaelscholle, 2022-04-11:
The clean builder from michi

Consti10, 2022-04-11:
> The clean builder from michi
> 
> I would like the milestone release to be build directly with our github builders, nothing just "uploaded"

That makes sense

MacDaddyFPV, 2022-04-12:
I have done the foundation for building raspi2png in drone.io but it is running into an issue I am yet to solve. It requires a file called bcm_host.h but it is not present in the image location as expected.

MacDaddyFPV, 2022-04-17:
gst-rpicamsrc has now built by updating to a raspbian based docker image rather than debian.

raphaelscholle, 2022-04-27:
looks like only QopenHD needs to be finished

Consti10, 2022-04-30:
Notes from discord (only german)

1) openhd-common
-> passt
2) openhd-power:
Methoden zum auslesen gemacht, custom mavlink messages dafür sind schon da
-> TODO: QOpenHD muss die neuen mavlink messages benutzen
--> Eigentlich alles ready zum löschen

3) openhd-security
Erst einmal deaktiviert lassen & keine encryption in wifibroadcast

4) openhd-status:
Kann weg, funktionalitäten voll in XTelemetryService

5) openhd-telemetry:
Ersetzbar, nur: Ich muss noch das zeug mit den default UART's von stephen reinkopieren.
& Default baud: std::string platform_baud = "115200";

6) openhd-tracker: Leiche von früher, kann erst einmal raus

7) openhd-video:
TODOS: Ports heruasfinden, als kontsanten definieren und hardcoden.

Andere TODOS:
Ich kopiere die richtige wifibroadcast lib als submodul unter lib/*
-> das ganze andere wifibroadcast zeug kann dann weg

overlay: wird mit in deb reingepackt & kopiert diese mit rüber.

raphaelscholle, 2022-04-30:
Notes from discord (translated)
1.	openhd-common
-> is ok
2.	openhd-interface:
Telemetry links (bi directional) start hard coded
Video links (one directional) start (tx in air, rx on ground)
3.	openhd-power:
methods to read are made, custom mavlink messages are ready
-> TODO: QOpenHD needs to use the new mavlink messages
--> everything can be removed now
4.	openhd-security
deactivated until needed -> no encryption in wifibroadcast
5.	openhd-status:
will be removed, XTelemetryService works
6.	openhd-telemetry:
works, but I need to copy UART-Stuff from stephen

& Default baud: std::string platform_baud = "115200";
7.	openhd-tracker: can be deleted, is not used
8.	openhd-video:
TODOS: find ports, define as constants and hardcode it
additional TODOS:
I copy wifibroadcast lib as submodule in lib/*
-> other versions of wifibroadcast can be removed
overlay: is added to deb & just copies them over
//just translated for our international people (just some dev-notes)

Consti10, 2022-05-01:
Putting this out there for the future:
For this release, we should make sure that we update the master branch with the code used in the release.
Also, we should clean up stale branches.

Consti10, 2022-05-15:
Notes from meeting on Discord:

Goal: build and install the OpenHD/consti-test code with the image builder
Goal finished: ./OpenHD can be executed on the image from image builder
( or also: ./OpenHD is automatically started on startup)

Difficulties: I've changed a lot regarding how to build OHD code (from services to single executable, different repository, ... )
Communication with Raphael might be required.
The CI test UbuntuX86Test shows how to build and install the project without the packaging step

Steps:
1) test a specific commit-hash on the rpi cm4 running "WithoutOpenHD" from the image builder
-> Test 1: Builds - yes
-> Test 2: runs -> yes (Executable-name: OpenHD , installed to bin and can therefore be executed anywhere)
Commit-Hash: 8611bd1b27330f1a784efe12a4cac887ddc4a80d

2) Let rapha do its magic. We'l look at this goal as done when the code compiles and can be executed, everything else will be tested / figured out afterwards

Consti10, 2022-06-03:
Since the question came up today, we should make this info publicly available:

OpenHD is always open to integrate new datalinks. To start this process, the following steps have to be performed:
# Steps to integrate new data links into OpenHD:
1) Make sure drivers for the link HW work in linux (RPI OS as a baseline)
2) Implement a simple pair of programs (tx and rx, look at wifibroacast as an examle) that follows this pattern:
UDP localhost -> air (tx)
air -> udp localhost (rx)
OpenHD does not "care" what device / hw is transmitting data, but to integrate a new HW solution with OpenHD, the above steps have to be tested and implemented.

Once the above steps have been performed, please contact OpenHD developers.

raphaelscholle, 2022-06-16:
I've finally managed to have QT working again on Pi, next step for me is updating Qt on the jetson to the newest version.

closed 2022-10-17 by raphaelscholle