OpenHD/OpenHD #633: How to safely deal with changing the wifi frequency

Consti10, 2022-04-20:
Changing the wifi card(s) frequency is REALLY HARD due to the two generals problem:
https://en.wikipedia.org/wiki/Two_Generals%27_Problem

Here is a solution that I would propose, that also catches a couple of common user issues:

1) We have a wifi service that runs on startup (before anything else). All services that use wifibroadcast (Telemetry,video) must run 
after this service has finished.

2) We define some assumptions that are true after the wifi service has run that make the development of services that run after it easier.

I'd define the following assumptions:
a) There is at least one wifi card, called wlan0
b) If more than one wifi card is connected, they are called wlan1, wlan2 and so forth
c) The wlan card(s) are in monitor mode
d) If the service runs the first time, it selects a (pre-determined) frequency channel for 2.4 / 5.8ghz

And a possible implementation could work the following:
1) Look for one or more wifi cards. If no wifi card is found, flash the RPI LED in 1 second intervalls and never return.
=> This way users can easily find the most common mistake ;)

2) Read the wifi frequency from a file. If this file does not exist (first startup), create a new one and write the default parameters.
Set the wifi card(s) into monitor mode.If this fails, flash the RPI LED in 2 second intervalls and never return.
=> Also a common mistake, e.g. connecting a wifi card that is not compatible (cannot be set into monitor mode) is catched.

3) Done - now if the user wants to change the frequency, we do the following:
a) send the frequency change command to both the ground and air station via extended messages. This is blocking, until the change has been performed. However, we only write the new frequency into the file. The actual change of frequency is only performed at reboot.
b) once we have acknowledgements from both the air pi and the ground pi, it is safe to reboot them both.

pilotnbr1, 2022-04-20:
This all sounds good to me

Minor detail is that openhd-interface wifi.cpp handles wifi cards based on their mac ids. Not wlan0.. There might be an exception to this with the jetson though (maybe x86) as I have seen in the jetson video bash script that attempts to bandaid some other issue (it always refers to the card as wlan0)

2-C not sure on intial start of the system that the cards should be in monitor mode. 
A proposal would be to have the cards behave as traditional wifi. 
1) 1 side (air or ground) has its card setup as a wifi AP
2) the other side (opposite air or ground) joins the network. The advantages of this is that a secure connection wifi connection can be made using a user defined password protected network
3) a discrete key can be exchanged to facilitate encryption or maybe just a random number for all future packets to contain.
4) the ground side requests the air to send settings. This can be done by extended protocol as well. Or perhaps just share a settings file as mavlink has provisions for file sharing.
5)when this "sync" is complete the ground signals the wifi connections to be reset to the user defined settings(freq etc) and the system continues in the traditional wbc monitor mode data exchange (as proposed in 2-D )

pilotnbr1, 2022-04-20:
The acking of the requested freq change is onbviously important and I assume is handled by mavlink. We dont want air being commanded to change freq and the message never gets there  or the ack on the ground is not received. 

Basically we must avoid the situation where ground and air end up on different freqs.

We might also think about the situation where one side has an unexpected reboot. We want the rebooting side to immediately go to the last freq and regain connection as quickly as possible

Consti10, 2022-04-22:
> This all sounds good to me
>
> Minor detail is that openhd-interface wifi.cpp handles wifi cards based on their mac ids. Not wlan0.. There might be an exception to this with the jetson though (maybe [...]

Hm, my intention with wifi0 ... was that then one can just run the wfb instance(s) with wifi0 as parameter (without reading any json usw). I think some cards show up as wifi0, ... on some systems, while others show up as some cryptic numbers (like "wlx244bfeb71c05" ). As far as I know, one can just re-name them - but maybe it is better to write files called wlan0.json ... that contain the mac address.

> 2-C not sure on intial start of the system that the cards should be in monitor mode. A proposal would be to have the cards behave as traditional wifi.
> 
> 1. a side (air or ground) has its card setup as a wifi AP
> 2. the other side (opposite air or ground) joins the network. The advantages of this is that a secure connection wifi connection can be made using a user defined password protected traditional wifi network
> 3. a discrete key can be exchanged between air/ground to facilitate encryption or maybe just a random number for all future packets to contain.
> 4. the ground side requests the air to send settings. This can be done by extended protocol as well. Or perhaps just share a settings file as mavlink has provisions for file sharing.
>    5)when this "sync" is complete the ground signals the wifi connections to be reset to the user defined settings(freq etc) and the system continues in the traditional wbc monitor mode data exchange (as proposed in 2-D )

Not sure if doing so is worth the effort - there are 3 things that speak against such a quite complicated extra step:
1) It makes the requirements on the wifi card(s) features higher - I am not sure if every driver (especially our patched wifibroadcast drivers) support an AP mode feature.
2) We can easily do this with just wifibroadcast, and I'd guess the time to implement is the same as if we go with the hotspot "extra" (in dev time)
3) We then need to make sure that this feature does not extend the bootup time, especially after a crash (for example).

pilotnbr1, 2022-04-22:
> Hm, my intention with wifi0 ... was that then one can just run the wfb instance(s) with wifi0 as parameter (without reading any json usw). I think some cards show up as wifi0, ... on some systems, [...]

I mispoke slightly on card names. They arent mac ids on some platforms they are are alpha numeric "cryptic" ids. This is all done in openhd-system wifi.cpp :
boost::filesystem::path net("/sys/class/net");
    for (auto &entry : boost::filesystem::directory_iterator(net)) { 
        auto interface_name = entry.path().filename().string();

> Not sure if doing so is worth the effort - there are 3 things that speak against such a quite complicated extra step:
>
> 1. It makes the requirements on the wifi card(s) features higher - I am not [...]

As far as the setting up an AP stuff I started that a month ago in my own "setting-sync" branch. I successfully had one side setting up an ap. I think I left off with having the other side join. If it interested you or you feel its worth it- here is the bulk of my code https://github.com/pilotnbr1/Open.HD/blob/settings-sync/openhd-interface/src/sync.cpp

careyer, 2022-04-23:
I like both your ideas (Consti, yours but also Luke’s).

Just to throw this in for discussion: wouldn’t it be desirable to have the possibility to change WiFi frequency on the fly? I mean literally in-flight without reboot.
This should be possible with at least two cards connected with your ground station.

1) ground station commands Air unit to change frequency from freq_X to freq_Y.

2) ground station switches one off it’s two cards to the new freq_Y whereas the other card stays at the current freq_X.

3) Once the ground station receives a signal at freq_Y it knows Air unit has successfully switched to the new frequency and can also switch over its other WiFi card to the new frequency now. 

This works like a handover in real Air traffic control I guess 😂🤣.

This way the user could switch to a new frequency if his current frequency gets congested or if he suffers bad interference. 
This could also be automated to always choose the best channel available. 

Just thoughts.
Cheers!

Consti10, 2022-04-25:
> 

I think making a second wifi card a requirement just for changing the frequency is a bit overkill.

However, we can also go for a method that works roughly the same - after changing the frequency, having a timeout and listening for data. When no data comes in after the timeout, go back to the original frequency.

Consti10, 2022-04-25:
The first part I described (looking for the connected wifi cards) is already done in 
https://github.com/OpenHD/Open.HD/tree/master/openhd-system

(as far as I understand).

Making at least one connected wifi card that supports injection / receiving a requirement & flashing the error code should be easy to implement there, too.

Consti10, 2023-01-28:
Current approach is not perfekt by any means, but with the channel scan feature you can recover in case anything goes wrong.
(NOTE: Assuming both air and ground are using the same hardware / support the same channels)

closed 2023-01-28 by Consti10