OpenHD/OpenHD #658: building image for other small computer -Radxa Zero

massivedynamics-sd, 2022-07-22:
Dear developers
Honest greetings and respect 

After the global crisis and the lack of chips, it was difficult for us to provide Raspberry Pi for training. We turned to alternative systems such as Radxa Zero
Since I am new to  Open HD, please help by explaining how to download Open HD on Radxa Zero computer running debian-buster
https://wiki.radxa.com/Zero

Thanks a lot in advance 
Regards

raphaelscholle, 2022-07-24:
OpenHD does not support Radxa Hardware, so it can't be installed.

massivedynamics-sd, 2022-07-24:
dear [raphaelscholle](https://github.com/raphaelscholle)

> OpenHD does not support Radxa Hardware, so it can't be installed.

Dear raphaelscholle
thanks alot for your replay 
what if we recompile the source code ?

raphaelscholle, 2022-07-25:
Recompiling isn't the problem. We need a custom kernel, patched drivers, new decode/encode pipelines and additional stuff to get support in upcoming releases. 2.0x is not build to be used on non-pi hardware.
2.2 will enable more SBC's but radxa is not planned. And won't be added in near future, since there are no advantages over Hardware we already support. And adding additional Hardware costs money and time.

massivedynamics-sd, 2022-07-28:
> r [raphaelscholle](https://github.com/raphaelscholle)

Dear @raphaelscholle 
thanks alot for your kind replays
please check this hardware comparison with ability to transmit   H.265 to reduce needed communication bandwidth

massivedynamics-sd, 2022-07-28:
@raphaelscholle 
https://www.cnx-software.com/2021/11/01/raspberry-pi-zero-2-w-vs-radxa-zero-features-and-benchmarks-comparison/

raphaelscholle, 2022-07-28:
I know that some Rocklship boards are able to encode h265, thats why our custom hardware is using a rockship soc. 
The thing is, that the next releases will include the jetson nano, which has an really good h265 encoder. 
In addition to that the custom hardware which will be released in the future has some very nice ways to even reduce latency, but this function can not be used with off the shelf hardware.

That means that the Radxa will be less fast then the custom board, but about as fast as the jetson. It's just to many work to support hardware which isn't really better in anything.

If someone would sponsor the dev team with some boards we could look into that, but currently there is just not the time to integrate a new SOC.

massivedynamics-sd, 2022-07-28:
Dear @raphaelscholle 
I enjoyed very much the dialogue and discussion with you because it is easy and full of information.. and thank you very much for your responses to me.. and I am waiting for the new hardware..God bless you

massivedynamics-sd, 2022-07-28:
in the new hardware please include arducam camera driver 
**IMX462 2MP Low Light Camera Module**
 V4L2 Pivariety driver
**arducam 64MP cameras**
wget -O install_pivariety_pkgs.sh https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/install_pivariety_pkgs.sh

raphaelscholle, 2022-07-28:
We're working directly with arducam to include their drivers 
@low light, we already support some low light cameras like http://wiki.veye.cc/index.php/VEYE-MIPI-290/327_index

raphaelscholle, 2022-09-05:
Since Radxa is sending me a cm3 I'll see how easy it'll be to make it work as air-unit, so maybe we'll support it as air in upcoming releases, but it'll cost some time

massivedynamics-sd, 2022-09-05:
yes please god bless you

raphaelscholle, 2022-10-04:
To give a little update, I've managed to run openhd and use the rtl8812au card, but their documentation is lacking even the basics.
I needed to write with the support of radxa to get usb to work, and never got any camera recognised. 

Maybe I'll continue testing in the future, but for now it's not worth the time

buldo, 2023-01-21:
@massivedynamics-sd while experementing I wrote an instruction how to build OpenHD for radxa zero https://gist.github.com/buldo/5ef7fa562296ba341355bae926a2091e  
But there is only software decoder available.
If you have radxa-zero and want to use it, I can explain how. But this SBC is completly unsupported by OpenHD team. So it will be completly not tested.

raphaelscholle, 2023-04-15:
we've decided to include the Radxa Rock5 as officially supported platform in the upcoming releases, when Radxa updates the CM3-Kernel I can look further into the CM3 family, but till this happens there is no development happening on the cm3.

closed 2023-04-15 by raphaelscholle