OpenHD/OpenHD #735: Error: boost::filesystem::directory_iterator::construct: No such file or directory: "/sys/class/net"

fuchen9475, 2022-11-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration and debug logs**
Please attach the relevant openhd-settings*1.txt and any debugging logs available

**Hardware:**
 - Air side
 -- Pi model
 -- Camera(s)
 -- Wifi card(s)
 -- Other hardware
 - Ground side
 -- Pi model
 -- Wifi card(s)
 -- Ground Control Station(s)
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?

**Client devices (in case of hotspot/tethering issue)**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App - FPV_VR, QGC, and version
 - Version [e.g. 22]
 - Connection method

**Additional context**
Add any other context about the problem here.

Consti10, 2022-11-13:
hm, never seen that before / can you explain a bit what you are doing ?

Could be that the wifi card detection code doesn't check if there is a directory called /sys/class/net and creashes if i doesn't exist.

I'd think every linux installation should have this directory though, right

fuchen9475, 2022-11-14:
I am doing this in my phone(oneplus6),
I installed App called termux ,and installed debian 11 with termux ,and compiled OpenHD ,
It was wrong with "Error: boost::filesystem......" while I was running openHD.

Consti10, 2022-11-15:
OpenHD cannot be run in a VM, since it needs to directly talk to the wifi hardware.

closed 2022-11-15 by Consti10