OpenHD/OpenHD #804: OHD evo request for features

Flierman1945, 2023-01-27:
1 Can we have a facility to start and stop video recording easily. If the air vehicle is standing idle, video records (if set) and you end up with loads of unwanted videos.   
2 Can we have a tab on the Air  wifi hotspot web gui to delete unwanted video files

Consti10, 2023-01-28:
1) is easier said than done, since starting and stopping recording is right now tied to (re)starting / stopping the camera stream.

But we are working on a solution.

knputyi, 2023-01-28:
Isn't it would be easier to bring up a widget on osd, that pushed has equal effect with the option of enabling / disabling video recording in Qopenhd?

raphaelscholle, 2023-01-30:
Widget:

- [x ] in work

- [x ] pull-request

- [ ] building

- [ ] testing

- [ ] release

Flierman1945, 2023-01-31:
Hi, Just curious as to what option have you gone for ref start stop video recording

knputyi, 2023-01-31:
> Hi, Just curious as to what option have you gone for ref start stop video recording

It's in work /test, as a widget with two sliders, "start rec" and "stop rec", also an icon green / red for confirmation and a free space counter.

Flierman1945, 2023-01-31:
Just an observation.  Will this require a touch screen or a permanent mouse connected to allow this operation of start stop

raphaelscholle, 2023-01-31:
Currently all widgets require a mouse or a touchscreen

Flierman1945, 2023-01-31:
Yes, For those who don't have a touch screen, a mouse permenantly connected can be a bit problamatic. Forgive my ignorance, but is it posible to pick up an arming signal from mavlink and use that to start ; stop video record.  I appreciate this may just not posible.

raphaelscholle, 2023-01-31:
Possible, yes, but a lot of coding needed, we can add that when we're working on rc-interactions, but that's a pretty low-priority right now to be honest.

We can't just use the same method like in the widget, because the settings are lossy (sometimes applying them fails and needs to be done again).

Flierman1945, 2023-01-31:
Thanks for the reply  Rapha, appreciated

Consti10, 2023-02-01:
> Yes, For those who don't have a touch screen, a mouse permanently connected can be a bit problamatic. Forgive my ignorance, but is it posible to pick up an arming signal from mavlink and use that to start / stop video record. I appreciate this may just not be posible.

similar to adjusting tx power when drone arm / disarm. Doable, but requires some critical coding, so r.n not on the priority list.

Consti10, 2023-04-02:
Fixed

closed 2023-04-02 by Consti10