OpenHD/OpenHD #817: Problems changenig frequency, mcs....

Droneforest05, 2023-02-05:
I'm using a raspi 4 on land with an alpha 036ac
and in air raspi 3 b+ and ac56
another problem that I think another user in the group has already commented on.
The frequency changes, mcs, are not done correctly, sometimes it changes well, but other times when air and ground are connected and doing a fetch, for example, you change the frequency (from 5280 to 5180) and it only changes in ground and you get a message that It has not been able to air because it is already at 5180¡¡ and of course you lose connection and you have to look for the frequency.

but more serious is if, for example, you try to test mcs settings and that happens to you as it happened to me when I change it from mcs3 to mcs4 and it only changes you on the ground because the air has rejected the change and this leads to the air being disconnected and land and you have to land the uav without video because also you can not change it back to mcs3 because there is no connection it does not let you make any changes

https://user-images.githubusercontent.com/110037180/216837936-915b7db8-898d-4797-9c8e-39ef14b646d9.mov

Consti10, 2023-02-05:
Related https://github.com/OpenHD/OpenHD/issues/816

Consti10, 2023-02-05:
could be that it is related to this from the release notes:
changed openhd settings path (camera, interface,NOT QOpenHD) - A lot of (rpi) users complained that they cannot change settings manually anymore.
Even though this is not recommended, we want to support that - and since on rpi image only /boot shows up
under windows in the file reader, we had to change the path in this regard. Shouldn't create any issues
on linux, since we are root, we can just cretae the directory at run time

Will investigate.

In the meantime, changing frequency and/or mcs index is not recommended during flight - they are more intended to be set once according to preferences, then left undtouched during flight

Consti10, 2023-02-06:
Looks like there is a pretty serious bug that slipped under the radar.

The persistent storing of settings (both air and ground) has been compromised by changing the path.

In short, it looks as if I power down / restart any rpi by unplugging power (which is quite common, no one properly shuts down its air / gnd pi ;) all persistent settings can be lost and OpenHD then thinks it is a fresh start (since it has nothing to read from) and applies default settings.

Fix should be easy (pretty much one line of code) but I gotta double check before we can update the released image.

Consti10, 2023-02-06:
https://github.com/OpenHD/OpenHD/commit/6ee96d5572e1ddc332bf5c18b726b3309260b37b

Droneforest05, 2023-02-06:
Is there going to be a new release to fix this or are we going to wait until we can include something else?

Consti10, 2023-02-06:
Since it is quite a serious bug, we'l create a hotfix just for that.
https://github.com/OpenHD/OpenHD/releases/tag/v2.3.1-evo-beta-hotfix1

To complete this issue:
Watching your video closely, I think you lost your video during flight not while changing the mcs index, but rather the channel width. Just like with other solution(s) (like dji fpv v1) we do not recommend to either change frequency or channel width during flight. To make sure this doesn't happen by accident, we should add a warning if the drone is armed & the user wants to change these 2 params.

The mcs index should be changeable during flight without problems, since it doesn't need to be synchronized.
That's how we plan to add variable bitrate at some point.
(The big issue here is that so far, only rtl8812au allows changing the mcs index properly)

Droneforest05, 2023-02-06:
I don't know if the problem in the video is due to changing the "width" but today I did a test at home (I forgot to record it tomorrow I'll repeat it) I tried to change from mcs 3 (default) to 4 (I don't know if 5.8 supports it) to see how it would look and at the moment it gives me the message that there was an error in the air and that it cannot make the change after that the image freezes but the telemetry continues to work, it cannot be changed again because air rejects any change giving an error the only thing that can be done is reflask the sd

Consti10, 2023-02-07:
There are 2 ways to fix a frequency mismatch - find air unit or placing a file reset_frequencies.txt under /boot/openhd

Weird though, since if I understand you correctly the issue happened when chaning the mcs index. I am still not sure mcs index needs to match or not, or perhaps there are some eigentlich cases where it needs to match.

Consti10, 2023-02-07:
Sorry, reset_freg.txt under /boot/openhd/

Droneforest05, 2023-02-07:
I think we are confusing concepts, what I was referring to in the previous message was the mcs all the time, here is a sample of what I am saying in this case it does let me change it but then it stays kind of crazy and it only goes one frame every 5 seconds or something like that but the telemetry works because the link is still there but since it doesn't let you make changes because the air unit doesn't want to and I think this was what happened on air.
I had to upload the video to youtube because it is very big I leave you the link
https://youtu.be/oRg197kNSBE

Consti10, 2023-02-07:
Is it possible you were using a tx power index higher than high with those Tests ?
A higher mcs index means more advanced Modulation, and depending on your setup this might result in an overloaded tx rf circuit.
Take a look at your packet loss in percent in the upper left corner - if it is higher than like 2-5% on a specific mcs index & tx power index your tx power index is probably too high. Aka the tx is producing a lot of bad packets due to overamplified tx circuit.
We are not 100% sure on that yet, but I think that's the case.

Consti10, 2023-02-07:
Regarding the 'rejected' message - it is a bit misleading, we do not differentiate between rejected (this would be always the case if your tx for example is not rtl8812au) and simply "couldn't change it with X retransmissions due to too high power index"

Droneforest05, 2023-02-07:
in air i have tx power 58 but this is the value that use in 2.0(actually in 2.0 i use 60) and now looking to the packet loss is 1% with default config (mcs3, tx power 58 and width 20 mhz)

Droneforest05, 2023-02-07:
and now that I remember I still get the x failed polls message even when I have a tracker connected, I don't think it's very important but it's strange

Consti10, 2023-02-07:
> 

I don't think the mcs index was really a thing in 2.0 (especially anything higher than 3) so that's probably why you were fine with higher power Indices with it.
Like I said, mcs index and Power index need to be adjusted (and unfortunately this stuff is unique per setup, even unique per Adapter- like overclocking a cpu )
There is also a report by svpcom about some users reporting issues even on 45 as tx power index (probably depends on Wiring and input voltage then)

Consti10, 2023-02-07:
> and now that I remember I still get the x failed polls message even when I have a tracker connected, I don't think it's very important but it's strange

Yeah, can be safely ignored, it means the gnd rpi uart is trying to read something from the tracker uart but the tracker doesn't give anything out so to say. (Assuming you set your tracker to the 115200 baud default on the gnd uart)

Droneforest05, 2023-02-07:
> Yeah, can be safely ignored, it means the gnd rpi uart is trying to read something from the tracker uart but the tracker doesn't give anything out so to say. (Assuming you set your tracker to the [...]

ok yes the tracker is connected and working and set baud to 115200 no maybe my tracker (diy) is not given this answer to the rpi

Droneforest05, 2023-02-07:
> I don't think the mcs index was really a thing in 2.0 (especially anything higher than 3) so that's probably why you were fine with higher power Indices with it. Like I said, mcs index and Power [...]

ok so i left it in 3 and it should be fine, but I honestly believe that this should be improved or at least warned because I am very nosy and big-handed and I know that what I play can screw something up but many people do not have much idea that this should not be played so lightly (without some warning or something) he touches it, he gets screwed and they leave ohd (to ruby for example) because it is not very user friendly
I think that these things should be put in a tab for example called (advanced options) and a warning

Consti10, 2023-02-07:
the air recording is now explicitly not saved - or rather disabled at boot ;) such that you don't fill up your storage by accident.
With the widget, this makes more sense in my opinion.

Consti10, 2023-02-11:
Just letting you know we are actively working on those frequency issue(s). There are a couple of things that can lead to bugs, for some part, they come from the flakiness of the WIFi drivers, for the other part, they are bugs in code.

Droneforest05, 2023-02-11:
Ok no problem i not have time to test the new versión well i Hope tomorrow ill have time

Consti10, 2023-03-18:
Those problems should be fixed in the last release

closed 2023-03-18 by Consti10

Droneforest05, 2023-03-18:
Nice