OpenHD/OpenHD #840: INA219 power monitoring

Consti10, 2023-02-25:
We now have had a couple of users asking for it.

1) What is INA219 ground power monitoring: Commonly used by people with custom built ground stations, connect a INA219 via i2c to the ground pi and the batterie(s) on the ground pi

2) Why can't we just copy paste code from 2.0 / pre-evo:
the code / scripting that provided this functionality is not well documented & uses outdated libraries

3) Recommended steps to reintroduce it:
a) Make sure no other feature uses the ground pi i2c -> check
b) Either implement reading & forwarding the values in a custom service specific for ground pi or add the functionality in OpenHD under ohd_telemetry (see [example](https://github.com/OpenHD/OpenHD/blob/2.3-evo/OpenHD/ohd_telemetry/src/internal/OnboardComputerStatusProvider.h#L15) for something similar
c) Either integrate the value(s) into an already existing mavlink message or define a new one if needed
d) parse and display the values in QOpenHD

raphaelscholle, 2023-06-04:
initial power monitoring is included in the newest release, there might still be issues with it, so we would be happy if somebody tests/reports issues

raphaelscholle, 2023-06-14:
finally had a little time to check it, it is calculated correctly on the ground, maybe the debug message should be removed, since it is quite annoying coming each second, it's also visible in the power tab in QOpenHD, but not in the power widget right now

raphaelscholle, 2023-06-19:
ground monitoring is again usable

closed 2023-06-19 by raphaelscholle