OpenHD/OpenHD #855: Open HD Support for PCIe based card

abhi11796acharya, 2023-03-18:
Hi,

First of all, amazing project.

I am new to open HD. I have gone through the wiki page and it seems like it support till now only USB based cards. I am looking to use it with Jetson platform which has PCIe bus.

I am curious that any one know any PCIe based cards which open HD supports?

Consti10, 2023-03-18:
2 things to note here:
1) jetson support will be dropped in the future and replaced with RK3588 support (there are multiple reasons to do so)
2) In general, USB driver(s) are easier to use and much more versatile than PCIe driver(s). So I highly recommend using a rtl8812au usb card, no matter which platform you are running on

raphaelscholle, 2023-03-21:
if you use a 8812au pcie card it should also work, but won't have any advantages

raphaelscholle, 2023-05-19:
The Jetson is discontinued and Nvidia never bothered to update their kernel at all, it's simply ancient

raphaelscholle, 2023-05-19:
The Jetson is discontinued and Nvidia never bothered to update their kernel at all, it's simply ancient we even need to compile GCC from source to get anything to work

raphaelscholle, 2023-06-08:
let's just close this with "it doesn't really matter how you connect a card, if the chipset works with openhd it'll work even as pcie card"