OpenHD/OpenHD #862: Project Update Please

pilotnbr1, 2023-03-28:
Hi guys! I am still alive- just had some telegram issues so havent been around. If not too much trouble would you be able to provide a brief update on progress? Thanks! As an outsider trying to look in without telegram, the project progress is very opaque :)

pilotnbr1, 2023-03-28:
In discord I came across this update it answers some questions for me. Ill post for anyone else following via github

--------------------------------------------------------------------------------------------------------------------------------------- 

OpenHD evo 2.3.3-beta

We are pleased to announce OpenHD evo 2.3.3, which includes bug fixes and new features based on user feedback.

Bug fixes:

Variable bitrate: Fixed bug where the video bitrate was unneccessarily decreased too much
force_sw_encode not persistent during reboots (usb cam)
in multi camera mode, the curr recomended bitrate reported by the mavlink message(s) is wrong. Internally, the right one is used though.
RPI decode: Fix bug where lower part of the image can be corrupted in some cases
MCS index - only change on air unit (downlink)
Fix RSSI percentage value not shown
Fix flight mode command showing error message (though successfull)

New features:

Improve feedback / visual indication(s) what's going on with the bitrate(s):
a) Show a message in the HUD if the openhd recommended encoder bitrate changes
b) If there is a high difference between the set an measured encoder bitrate, the video bitrate widget text is colored red
Expose STBC,LDPC and GI as dynamically changeable params (advanced users only)
add option to force sw encode also for rpicamsrc and libcamerasrc
Add option to show FC message(s) in HUD (default on, can be disabled)
Improved/more documented settings
Improved image quality on arducam lowlight cameras

pilotnbr1, 2023-03-28:
I do see some commits regarding jetson. Any h265 progress?

Consti10, 2023-04-02:
Hey,

I do agree, for an outstander the progress might not be as opaque as we'd like, this is mostly because a lot of decisions and updates for developers are done on the weekly developer meetings in discord.

Without joining them, one might not totally understand what's going on.
The git commits, releases and release notes should give a okay understanding already though.

E.g. In regards to h265, we support it since probably like half a year, only proper hardware is missing.

closed 2023-04-12 by Consti10