OpenHD/OpenHD #864: Developer Discussion - RTP

Consti10, 2023-04-02:
Putting this one out here for discussion -

should we go back from using rtp exclusively for the transmission from air to ground to a different, more simple approach 
(e.g. manually fragmenting frames on the air unit for FEC, and re-assembling them on the ground and then packetizing them into rtp).

There are a couple of reasons to do so:
1) Memory and libs on embedded devices: The progress of rtp packetization might sound simple at first, but you pretty much require a library like gstreamer to do so, and even then it can be quite annoying. With keeping the rtp on air paradigm, we'l have a hard time getting rid of gstreamer on air.

2) CPU performance is limited on the air unit (where rtp packetization happens r.n). While this is 0 issue on SOCs of a performance level like RPI 3 or similar, other embedded devices like allwinner do require more careful dealing with resources.

3) Since we now use callbacks for the interface between ohd_video and ohd_interface / wifibroadcast, we are not limited by the UDP max packet size and the neccessity to use RTP that arose from it

4) We'l save some minimal data overhead on the link, but this one is more of a overkill optimization.

Consti10, 2023-04-02:
Does this affect the support for h264,h265 and mjpeg: Not really, h264 and h265 can be safely dealed with using NALU byte stream, and generally still have the context of an existing "frame". MJPEG is frame per frame anyways already.

whoim2, 2023-04-02:
https://github.com/OpenIPC/silicon_research

This one teorethic way for use at ip cam and nvr from hisilicon. Vencoder and vdecoder - hw fastest "devices". At this example, rtp264 and NAL fragmentic, its work with wfb.

Consti10, 2023-04-02:
Interesting, so you also decided to do some simple custom fragmentation / refragmentation and not use rtp ?

Regarding your message in the chat:
I think, qopenhd need two variants, with rtp and without, changed in videosettings. As in fpv-vr

If possible, I'd like to avoid that, since it keeps things more standardized to only use rtp (and on the gnd, with optional forwarding via etherent and more, but without fec and things, rtp suddenly becomes much more feasible)

whoim2, 2023-04-02:
> Interesting, so you also decided to do some simple custom fragmentation / refragmentation and not use rtp ?
>
> Regarding your message in the chat: I think, qopenhd need two variants, with rtp and [...]

I didn’t decide to do it, but there is such a code example in sdk hisilicon )
The fact is that on devices like nvr there are few resources and additional conversion is a problem, gstreamer is missing, ffmpeg is only ancient and in general I don’t want to load it with an extra conversion. Then let rtp come from the air, the camera with the streamer forms it without problems.

closed 2023-08-22 by raphaelscholle