OpenHD/OpenHD #874: ohD 2.3 evo no Mavlink over Wifi Hotspot

wolkstein, 2023-04-18:
**Describe the bug**
No Mavlink over Wifi Hotspot

**To Reproduce**
Steps to reproduce the behavior:

1. Configure Fc Uart Connection on Air SOC
2. Check Telemetrie Infos in Main Screen (QOpenhd)
3. Enable Wifi Hotspot
4. Connect to Wifi Hotspot
5. Start QGroundcontrol on Client Pc connected to Wifi Hotspot
6. Optional, and default Settings in QGround (Enable Auto Connect UDP on Startup)
7. Or create your own Mavlink Connection in QGC Connection Tab (Type UDP, Default Port (14550)

Result, no connection and no Mavlink Data.

**Expected behavior**
I Expect that QGround automatically receive UDP Mavlink if in QGC Autoconnect is enabled, and an Wifi Connection to Openhad is activ.

**Configuration and debug logs**
Please attach the relevant openhd-settings*1.txt and any debugging logs available

I use 2.3 evo and on my Filesystem i can not locate an openhd-settings*1.txt

**Hardware:**
 - Air side 
 -- Pi model: Pi Zero 2 W
 -- Camera(s) Pi Cam
 -- Wifi card(s): RealTec 8812au
 -- Other hardware
 - Ground side
 -- Pi model: Pi 4
 -- Wifi card(s): RealTec 8812au, And Hotspot on Pi buildin Broadcom 
 -- Ground Control Station(s):
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link? I use RC over Mavlink Telemetry (Works well)

**Client devices (in case of hotspot/tethering issue)**
 - Device: X86
 - OS: Ubuntu Linux
 - App - QGC 
 - Version 4.2.4
 - Connection method: UDP over Wifi

ThX
Wolke

Consti10, 2023-04-19:
Yeah, we should probably mention that, wifi hotspot does not forward video / telemetry right now.
Mostly because I do not consider it a good practice to use it for that, since it creates interference during flight and is unsafe due to great packet loss. A better alternative (USB tethering) also exists.

You can do that anyways, though, by adding your receiving device IP to the manual forwarding ips in hardware.config (advanced users /  linux knowledge required).

Consti10, 2023-04-19:
I might have an easier solution though

yuace, 2023-04-20:
> You can do that anyways, though, by adding your receiving device IP to the manual forwarding ips in hardware.config (advanced users / linux knowledge required).

@Consti10 Now openhd default will forward the video stream,but didn't forward the telemetry data automatic when I connect usb tether?
[image]

Consti10, 2023-04-21:
Hm, it also should forward telemetry data, otherwise that'd be a bug (and there is some funkiness / bug I was not yet able to properly track down though) in regards to forwarding in some weird network cases.

Can you describe which network the ip is on you do manual forwarding to ?

yuace, 2023-04-21:
need manual forward?
[image]
You mean I need change this IP by change code?

Consti10, 2023-04-21:
No, hardware.config has an advanced setting for that. If you add a IP address there, OHD Ground will automatically forward video and telemetry to this ip.

yuace, 2023-04-21:
OK, get it. I will check the code and config setting.Thanks a lot. 
But seems like will listen the usb0 net and forward video automaticlly, right?
So video and telemetry use different strategies that's great.

Consti10, 2023-04-21:
yes, tethering is autmatically detected by openhd and then forwarding of video and telemetry to the ip of the tethered device is started

yuace, 2023-04-21:
> yes, tethering is autmatically detected by openhd and then forwarding of video and telemetry to the ip of the tethered device is started

You mean both video and telemetry automatically? But I just find the video automatically and telemetry didn't.

Consti10, 2023-04-21:
Yes, both. I just checked, in my case it worked. But if you can find out what's going on, that'd help

closed 2023-06-08 by Consti10

yuace, 2023-06-08:
收到！！！