OpenHD/OpenHD #897: No second camera view when attempting dualcam setup

easytidylanding, 2023-05-23:
I can't see or find the secondary camera view when attempting the dual camera setup

After working through the instructions here a couple times:
https://openhd.gitbook.io/open-hd/advanced-setup/dual-camera

I get the view of the primary camera, two camera icons that show two encoding rates in the top right of the QOpenHD screen, two options for recording via the top left camera icons, and when inside the OpenHD menu, I see correct 'Info' for the two cameras after fetching. However, I don't see any view, or way to view, the secondary camera stream.

**Hardware:**
 - Air side
 -- Pi model 4B, 4g
 -- RPi NoIR camera version 2.1, 
 -- Logitech C270 720p USB webcam
 -- Wifi card(s): dual TP3U

 - Ground side
 -- Pi model 400 (built into keyboard)
 -- wifi same as above
 -- Dell HDMI monitor

No other hardware involved, I am running this on a desktop setup prior to moving to a vehicle. Everything appears to be working...except the second stream.

Thanks for any help with your impressive project.

raphaelscholle, 2023-05-29:
sorry for the delay, I would guess that you only enabled dualcamera on the air and did not set up qopenhd to 2 cameras.
(for faster support I can suggest our telegram or discord chat's)

Consti10, 2023-05-31:
If you can fetch both cameras, and the bitrate(s) are showing on both camera icons, but there is no secondary video,
you are probably missing one of the required packages (but in 2.3.4). Try 2.3.5 instead ;)

easytidylanding, 2023-06-06:
Thanks for response - yes I was able to fetch, get bitrates and even record to SD - but not see the PiP.

I have tried to reinstall a new image but am now getting Qt problems. I'll assume the new version solves this and close.

closed 2023-06-06 by easytidylanding