OpenHD/OpenHD #926: Porting ground station to tvbox with armbian

Vittaly, 2023-06-28:
Good Day, all!

I wont to use tvbox (amlogic s905x3) as ground station for openHD.
I have done compile openHD and QOpenHD parts and start it (thank you for [chapter ](https://openhd.gitbook.io/open-hd/developers-corner/porting)in doc)

For testing I bought 2 wifi [dongle ](https://aliexpress.ru/item/4000817147623.html?spm=a2g2w.orderdetail.0.0.3de84aa6QdzHoe&sku_id=12000021324603402)with MT7601U and now see in system:

vit@armbian:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter

But now I have problem with OpenHD:

[2023-06-29 02:15:24.540] [main] [info] All OpenHD modules running
[2023-06-29 02:15:28.387] [wb_streams] [warning] Card wlan1 (might) not support injection/TX
[2023-06-29 02:15:33.391] [wb_streams] [warning] Card wlan1 (might) not support injection/TX
[2023-06-29 02:15:38.394] [wb_streams] [warning] Card wlan1 (might) not support injection/TX

How I think, I should to make patch for linux kernel and build driver ath9k_htc

But I dont know what changes in kernel I have to do. Can you help my with this problem? Thank you in advance!

raphaelscholle, 2023-06-30:
1. MT7601U  is not supported
2. ath9k is about the worst supported chipset, just go with rtl8812au
3. for ath9k you need a patched firmware, and patch cuda, the atheros driver to work correctly in injection, is's not that simple

Consti10, 2023-07-02:
> Good Day, all!
>
> I wont to use tvbox (amlogic s905x3) as ground station for openHD. I have done compile openHD and QOpenHD parts and start it (thank you for [chapter [...]

To fix a cards wifi driver, you need to patch its injection capabilities in monitor mode. You can check out rtl8812au driver for how to do that (warning: it is not simple). Let me know if you need more help

Vittaly, 2023-07-03:
@raphaelscholle  Thank you for your answer!

> 1.   MT7601U is not supported
But in the chapter [WiFi Adapters](https://openhd.gitbook.io/open-hd/hardware/wifi-adapters)  this chip is in the list of compatible chip. Is it mistake in chapter?
> 2. ath9k is about the worst supported chipset, just go with rtl8812au
Ok. I will try to work with rtl8812au (I ordered taobao)

Vittaly, 2023-07-03:
> To fix a cards wifi driver, you need to patch its injection capabilities in monitor mode. You can check out rtl8812au driver for how to do that (warning: it is not simple). Let me know if you need more help

If I install driver from repo https://github.com/OpenHD/rtl8812au, it will be not enought?

raphaelscholle, 2023-07-03:
Good that you noticed, it looks like this chipset was once supported in 2.0x and wasn't removed.
I can't really test it, but I would think this isn't supported, since I only heard bad things about the in-kernel drivers.

@8812au, it should work if you install this driver

Vittaly, 2023-07-10:
Hi all. I Found  wifi  adapter archer t4u plus with rtl8822

Now I see in log:

> [2023-07-11 01:20:22.233] [config] [error] Ill-formatted config file ini file not found.
> Failed to open the i2c bus: No such file or directory
> Failed to acquire bus access and/or talk to slave device: : Bad file descriptor
> [2023-07-11 01:20:22.274] [config] [error] Ill-formatted config file ini file not found.
> [2023-07-11 01:20:22.276] [config] [error] Ill-formatted config file ini file not found.
> command failed: Operation not supported (-95)
> command failed: Operation not supported (-95)
> command failed: Device or resource busy (-16)
> command failed: Device or resource busy (-16)
> [2023-07-11 01:20:23.606] [wifibroadcast] [error] unknown encapsulation on wlan2
> [2023-07-11 01:20:23.632] [wifibroadcast] [error] unknown encapsulation on wlan1
> [2023-07-11 01:20:23.726] [wifibroadcast] [error] unknown encapsulation on wlan2
> [2023-07-11 01:20:23.762] [wifibroadcast] [error] unknown encapsulation on wlan1
> [2023-07-11 01:20:23.799] [wifibroadcast] [error] unknown encapsulation on wlan2
> [2023-07-11 01:20:23.832] [wifibroadcast] [error] unknown encapsulation on wlan1
> [2023-07-11 01:20:23.834] [config] [error] Ill-formatted config file ini file not found.

Is it critical errors or not?

Consti10, 2023-07-10:
> Hi all. I Found wifi adapter archer t4u plus with rtl8822
>
> Now I see in log:
>
>
> Is it critical errors or not?

to me, that looks quite "bad" - have you confirmed the card properly does monitor mode with injection ?
Like already said, rtl8812au (exactly the code) is the way to go

Vittaly, 2023-07-10:
@Consti10  You are right :(

> [2023-07-11 01:55:24.545] [default] [debug] Card wlan1 reports driver:rtl88x2bu supports_2GHz:true supports_5GHz:true supports_monitor_mode:false supports_injection:true

Consti10, 2023-07-10:
the bu driver can do monitor mode and injection, if you install the correct driver - however, even this driver is far from fully functional. I highly recommend rtl8812au -  especially if you are new to openhd.

raphaelscholle, 2023-07-11:
I would really recommend using this driver : https://github.com/OpenHD/rtl88x2bu
it is tested and works with openhd quite well.

Just keep in mind that the bu can not do a lot of the fancy things the au card can do like 40mhz bandwith or high power transmission.

Consti10, 2023-07-11:
And it doesn't support changing the rate, mcs is fixed.

closed 2023-08-22 by raphaelscholle