OpenHD/OpenHD #934: I cannot download OpenHD

Genya55, 2023-07-09:
When I attempted to download "OpenHD-2.4.1-evo Raspberry" from "https://openhdfpv.org/download/openhd-raspberry",
following error appeared...

This XML file does not appear to have any style information associated with it. The document tree is shown below.

I also tried to  download from OpenHD Imager v1.8.7, but it said "Error downloading OS list from Internet"

kitsune0n, 2023-07-09:
same troubles

raphaelscholle, 2023-07-09:
sorry for the downtime, we're currently experiencing problems with our download server.
Until this is resolved here are the stable images 

 https://github.com/OpenHD/OpenHD-ImageBuilder/actions/runs/5427492705

raphaelscholle, 2023-07-11:
Server issues are resolved, everything should work now

closed 2023-07-11 by raphaelscholle