OpenHD/OpenHD #938: I am planning to use OpenHD for my unmanned vehicle, and I have some questions here

ZhouYixun, 2023-07-16:
My unmanned car originally wrote a program to run on the Raspberry Pi, which controls the servo and motor connected to the Raspberry Pi by receiving HTTP requests. May I ask if this program can still be used after installing OpenHD? Can OpenHD receive other signals besides the digital signals transmitted in the picture for me to control other servo and motors connected to the Raspberry Pi? I am planning to use Lora transmission

raphaelscholle, 2023-07-16:
OpenHD is designed around a flight controller which handles all the controls. 
The pi is actually pretty bad in handling controls, so that even a simple 3$ Arduino will be more reliable.

Since OpenHD uses a "regular Linux" you can install a lot to it, but you're on yourself if something interferes. Also remember if something freezes your car will just go on and on and on without the possibility to control it.

If you want to use lora in addition to OpenHD I would recommend using a 3.3V arduino to handle this.  Btw. the pi does only have 2 ADC's so you will have issues with controlling more than 2 servos/motors.

For a "rover" I would go for a low budget flightcontroller and ardurover, it's so much more usable.

ZhouYixun, 2023-07-17:
Thanks for your reply, I will reconsider the design of my car

closed 2023-07-18 by ZhouYixun